greatscottgadgets/hackrf #1274: HackRF with GNURadio send signal at 3.0GHz ~ 5.8GHz

jaehoon1001, 2023-02-08:
### What would you like to know?

Hi, 
I'm trying to send signal at 5.8GHz.
In 2.4Ghz, it is possible to send signal with hackrf and gnuradio.
but, 3.0GHz ~ 5.8GHz, I can't send signal.
I'm using hackrf, ANT500, and Ubuntu 20.04.
Is it my antenna issues? or something?
help me 🥲

dmaltsiniotis, 2023-02-09:
@jaehoon1001 Can you elaborate on what you mean by "I can't send signal"? Is there an error message or some kind? Also, what are you using to send the signal, hackrf_transfer, GNU Radio, etc...

It's worth noting that the [ANT500's](https://www.greatscottgadgets.com/ant500/) specifications are listed as 75 MHz to 1 GHz. 5.8 GHz is well outside the antenna's operating frequency, so you'll need an antenna that operates in the frequency range you're working with.

jaehoon1001, 2023-02-09:
What I meant was that even if I sent a signal to 5.8GHz with GNURadio, I could not get anything with gqrx. As you said, it might be an antenna problem. After changing the antenna, I'll check more. Thank you.

martinling, 2023-02-09:
Note also that the [maximum transmit power](https://hackrf.readthedocs.io/en/latest/jawbreaker.html#transmit-power) of HackRF One is much lower at 5.8GHz than at 2.4GHz, so you may need higher gains on both the transmitter and receiver, as well as suitable antennas, to successfully receive a signal.

jaehoon1001, 2023-02-09:
@martinling Wow, Thank you so much! With your answer, I can receive the 5.8GHz signal with the current antenna.

martinling, 2023-02-09:
Great! Glad we could help.

closed 2023-02-09 by martinling