greatscottgadgets/hackrf #1278: Gnuradio signal generator

markl17, 2023-02-10:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

The signal that’s generated is not locking in. It’s discussed here https://ham.stackexchange.com/questions/21686/unstable-frequency-generator-using-gnuradio-3-10 and the problem is that portapack frequencie number doesn’t change yet the signal strength number changes appropriately 

### What are the steps to reproduce this?

Run this on gnuradio https://gist.githubusercontent.com/marcusmueller/360aa8d07baec0b25063f75e27cde02f/raw/8f97745e27fc007f161e2bede9144106cf664b7b/hackrf.grc

### Can you provide any logs? (output, errors, etc.)

https://www.youtube.com/watch?v=RSlSGCbadUk

straithe, 2023-02-12:
It sounds like you are having an issue with Portapack and not HackRF One. Have you tried discussing with those folks?

markl17, 2023-02-12:
they decided to unsubscribe me they are way to german like they belong to
nazi party

straithe, 2023-02-12:
I'm sorry but we aren't the right folks to help out with this issue.

closed 2023-02-12 by straithe

markl17, 2023-02-12:
can you explain why is my issue closed

straithe, 2023-02-12:
Your question is focused on Portapack which we do not create. This is the repository for HackRF One. We aren't the right folks to help with your issue.

markl17, 2023-02-12:
what are you talking about which issue is it is it signal generator

markl17, 2023-02-12:
it has nothing to do with portapack I started the whole thing without the portapack the signal is very unstable