greatscottgadgets/hackrf #1279: Latest HackRF One (R9) doesn't support PortaPack H2+R3?

phil-stumpy, 2023-02-11:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Able to flash HackRF One with PortaPack H2+ [firmware](https://github.com/eried/portapack-mayhem) and PortaPack display to function correctly.

### What outcome actually happened?

When I connect the PortaPack H2+R3 to the HackRF One and power it on, with the latest HackRF One firmware (2023.01.1), I am unable to get the display working. The screen flashes briefly before returning to a black screen. I have tried holding one of the PortaPack buttons down whilst powering on, but no matter which button I hold down I get the same result - the screen flashes briefly before returning to a black screen.

If I disconnect the PortaPack H2+R3 (since I am unable to select "HackRF Mode" from the display) and try to flash the HackRF One with the PortaPack H2+ [firmware](https://github.com/eried/portapack-mayhem), using the command `hackrf_spiflash -w portapack-h1_h2-mayhem.bin`, I receive the following error:

```
    File size 1004328 bytes.
    Checking target device compatibility
    Compatibility test failed.
```
I have tried this with both the latest nightly build and the latest stable release, both with the same result (though slightly different file sizes). The only way I can flash the firmware is to use the `-i` option to ignore the compatibility test. Even with the PortaPack H2+R3 firmware apparently installed I am unable to get the display working, no matter which button I hold whilst powering on.

I have also noted that, after installing the PortaPack H2+R3 firmware and disconnecting the device from USB, if I then attempt to reconnect the device to USB it is no longer recognised and the USB and 1V8 LEDs no longer light up.

Having tried the various different options available to me, to get the PortaPack H2+R3 working correctly with the HackRF One, I am wondering whether: a) a hardware change in the R9 build has broken compatibility with the PortaPack H2+R3, or b) a change in the way the firmware is structured/installed has similar broken compatibility. Any help/guidance you can provide to assist me in diagnosing and correcting the issue would be much appreciated.

### What operating systems are you seeing the problem on?

Linux (Ubuntu 22.04.1 LTS)

### What is the output of ```hackrf_info```?

```
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000010961dc2a3a3d4f
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00584765
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

PortaPack H2+R3

martinling, 2023-02-11:
We have a pull request open on the Portapack firmware repository to add support for the r9 hardware:
https://github.com/sharebrained/portapack-hackrf/pull/187

martinling, 2023-02-11:
To elaborate a bit further:

HackRF One r9 uses some different components to previous revisions, so up to date firmware is required. The firmware support is included in our [2023.01.1 release](https://github.com/greatscottgadgets/hackrf/releases/tag/v2023.01.1). Its firmware images support both r9 and the previous board revisions.

The hardware changes in r9 do not break compatibility with the PortaPack hardware, but because the PortaPack project includes its own alternative firmware for the HackRF, that firmware needs to be similarly updated to run on the r9 board. We have written and tested those changes, and they are included in the [pull request](https://github.com/sharebrained/portapack-hackrf/pull/187) already linked above. That PR is against the original PortaPack firmware repository - the same changes should be applicable to other forks with minor adaptations.

We're not in control of the PortaPack project so it will be up to the relevant maintainers to merge those changes.

For the time being, you should be able to build a working PortaPack firmware by checking out the PR branch (`mossmann/h1r9`) and building from there.

phil-stumpy, 2023-02-11:
That’s awesome! Thanks for getting back to me so quickly, and good to hear that I’m not going completely mad 😆

I’ll take a look at the pull request branch and see if I can get something working with the PortaPack Mayhem build, until they get a new official build that is compatible with the R9 build.

closed 2023-02-11 by phil-stumpy