greatscottgadgets/hackrf #1280: Hackrf One not recognised anymore by host,  DFU works but firmware upgrade fails

avhoboken, 2023-02-12:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

hackrf one functioning correctly / connect properly to host

### What outcome actually happened?

device is not recognised anymore by Windows 10 or Ubuntu. In normal mode no device shows up in device manager. In DFU mode is does show NXP LPC device.
Also when running lsusb I see it is recognised properly 
[image]

following the documentation I tried to recover the firmware by the below command but I guess the result is not ok:
dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu 
[image]

Next also tried to upload the latest firmware but board is not found:
hackrf_spiflash -w hackrf_one_usb.dfu
[image]

### What operating systems are you seeing the problem on?

Both Windows and Ubuntu

### What is the output of ```hackrf_info```?

Even in DFU mode the board is not found
[image]

### Are you using any third-party software?

n.a.

### Are you using any third-party hardware?

n.a.

martinling, 2023-02-12:
The message `unable to read DFU status after completion` is normal and not a problem.

Once that `dfu-util` command has run, the NXP device should disappear and the HackRF One should show up. What do you see from `lsusb` and `hackrf_info` at that point?

It's important you don't unplug or reset the device before running `hackrf_spiflash`, since the firmware is only running from RAM at that stage.

Also note that the file you want to write with `hackrf_spiflash` is `hackrf_one_usb.bin`, not `hackrf_one_usb.dfu`.

avhoboken, 2023-02-12:
and a copy of my board
[image]

avhoboken, 2023-02-12:
Well good news, I already tried a few times before creating this issue but just run dfu-util again and now suddenly it shows hackrf device instead of the nxp device. Hackrf_info shows the below info and device is working again. 
[image]
Thanks for your feedback !!

closed 2023-02-12 by avhoboken

martinling, 2023-02-12:
You still need to run `hackrf_spiflash` to write the `.bin` file. The firmware is still running from RAM (see serial number `RunningFromRAM` in the `hackrf_info` output) so it will be lost when you next reset the device.

avhoboken, 2023-02-12:
Thanks, all works. Now next step is to get my portapack h2+ working again but for now I'm happy the the hackrf is working again :)