greatscottgadgets/hackrf #1281: access denied

markl17, 2023-02-12:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

anytime I run the program that was genrated by gnuradio 3.10 and I have access denied I am using  ** rtl sdr source** it says 
Traceback (most recent call last):
  File "C:\Users\markl\Downloads\top_block.py", line 231, in <module>
    main()
  File "C:\Users\markl\Downloads\top_block.py", line 209, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "C:\Users\markl\Downloads\top_block.py", line 101, in __init__
    self.soapy_hackrf_sink_0 = soapy.sink(dev, "fc32", 1, '',
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
no other version of gnuradio is not running I am actually running it in cmd mode https://imgur.com/bco0mvO

### What are the steps to reproduce this?

#!/usr/bin/env python3
# -*- coding: utf-8 -*-

#
# SPDX-License-Identifier: GPL-3.0
#
# GNU Radio Python Flow Graph
# Title: Top Block
# GNU Radio version: 3.10.4.0

from packaging.version import Version as StrictVersion

if __name__ == '__main__':
    import ctypes
    import sys
    if sys.platform.startswith('linux'):
        try:
            x11 = ctypes.cdll.LoadLibrary('libX11.so')
            x11.XInitThreads()
        except:
            print("Warning: failed to XInitThreads()")

from PyQt5 import Qt
from gnuradio import qtgui
from gnuradio.filter import firdes
import sip
from gnuradio import gr
from gnuradio.fft import window
import sys
import signal
from argparse import ArgumentParser
from gnuradio.eng_arg import eng_float, intx
from gnuradio import eng_notation
from gnuradio import soapy
from gnuradio.qtgui import Range, GrRangeWidget
from PyQt5 import QtCore
import osmosdr
import time

from gnuradio import qtgui

class top_block(gr.top_block, Qt.QWidget):

    def __init__(self):
        gr.top_block.__init__(self, "Top Block", catch_exceptions=True)
        Qt.QWidget.__init__(self)
        self.setWindowTitle("Top Block")
        qtgui.util.check_set_qss()
        try:
            self.setWindowIcon(Qt.QIcon.fromTheme('gnuradio-grc'))
[... 161 lines cut ...]
    signal.signal(signal.SIGINT, sig_handler)
    signal.signal(signal.SIGTERM, sig_handler)

    timer = Qt.QTimer()
    timer.start(500)
    timer.timeout.connect(lambda: None)

    qapp.exec_()

if __name__ == '__main__':
    main()

### Can you provide any logs? (output, errors, etc.)

Executing: C:\Users\markl\anaconda3\envs\gnuradio\python.exe -u C:\Users\markl\Downloads\top_block.py

Traceback (most recent call last):
  File "C:\Users\markl\Downloads\top_block.py", line 231, in <module>
    main()
  File "C:\Users\markl\Downloads\top_block.py", line 209, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "C:\Users\markl\Downloads\top_block.py", line 101, in __init__
    self.soapy_hackrf_sink_0 = soapy.sink(dev, "fc32", 1, '',
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

straithe, 2023-02-13:
RTL-SDR is not a piece of hardware made by Great Scott Gadgets. You will want to talk to the RTL-SDR folks instead.

markl17, 2023-02-13:
My friend why are you trying to push me off I own hackrf nothing else

straithe, 2023-02-13:
I'm trying to make sure you get the help you need with the information you've given me. Command line GNU Radio, RTL-SDR source, and the other information you've given here and in your other issue all suggested concerns with hardware and software I don't have the skills to help you with. I still can't assist you even with the updated knowledge that you are trying to get GNU Radio command line code to work with a HackRF One and not RTL-SDR. For help with GNU Radio command line projects I suggest checking out https://www.gnuradio.org/

markl17, 2023-02-14:
I still dont understand no where in anything I have said mentions rtl-sdr here is my error message
[INFO] Opening HackRF One #0 10961dc2975534f...
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.4.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf airspy airspyhf soapy redpitaya
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107800; UHD_4.3.0.0-release
Traceback (most recent call last):
  File "C:\Users\markl\Downloads\top_block.py", line 243, in <module>
    main()
  File "C:\Users\markl\Downloads\top_block.py", line 221, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "C:\Users\markl\Downloads\top_block.py", line 148, in __init__
    self.osmosdr_source_0 = osmosdr.source(
                            ^^^^^^^^^^^^^^^
RuntimeError: Failed to open HackRF device (-1000) Access denied (insufficient permissions)

>>> Done (return code 3221225477)

straithe, 2023-02-14:
I still can't assist you with this issue. For help with GNU Radio command line projects I suggest checking out https://www.gnuradio.org/

dmaltsiniotis, 2023-02-18:
At the risk of beating the proverbial dead horse....

> I still dont understand no where in anything I have said mentions rtl-sdr here is my error message

You stated you were using an rtl-sdr source in your opening sentence:

> ... I have access denied I am using ** rtl sdr source** it says...

No where in the issue is there an actual error message saying that something failed related to a HackRF One, only stack traces which aren't particularly useful in this case. Additionally, the screen shot of the flow graph you posted doesn't actually contain any HackRF sources or sinks, further adding to the confusion. It's only by sifting though the GNU Radio-generated python code that it appears your flow graph is trying to pipe the signal received by an RTL-SDR directly into a HackRF One for transmit.

As stated earlier, the folks here (myself included) are **not** GNU Radio experts, and cannot help with GNU Radio related issues. However, if you have a HackRF One specific problem, I would suggest taking GNU Radio out of the equation. The first step to validating your set-up is by running `hackrf_info` from the command line and placing the output here. That will tell you a great of information about your setup and help isolate your issue and hopefully get you going again.

In addition to hackrf_info can you validate if you've used the HackRF on successfully on this device (appears to be Windows 11) before? With any other SDR software?

closed 2023-03-18 by straithe