greatscottgadgets/hackrf #1282: use multimon-ng

jules35, 2023-02-14:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

i try to use multimon-ng
i succeed with an other sdr classic key but with rtl_fm
can you help me to find a command for the hackrf?
does rtl_fm work with hackrf? or is it any command line soft i can use foir pipe to multimon-ng?
kind regards

### What outcome actually happened?

                               .

### What operating systems are you seeing the problem on?

debian

### What is the output of ```hackrf_info```?

                                         .

### Are you using any third-party software?

multimon-ng

### Are you using any third-party hardware?

                     .

straithe, 2023-02-14:
I do not know what multimon-ng is. Can you please provide the link to the project?

jules35, 2023-02-14:
its a command can read pocsag. with rtl_fm | multimon-ng for exemple.( with rtl-sdr key)
but i cant use rtl_fm with the hackrf. anf i want to find a command like rtl-fm for send to multimon-ng

https://github.com/EliasOenal/multimon-ng

⁣finaly is for use with pagermon project

dmaltsiniotis, 2023-02-15:
> does rtl_fm work with hackrf
No, rtl_fm does not work with hackrf. However, if chain a few clever commands together you can get multimon-ng to work with hackrf output. You will need to modify the command below with your specific sound card settings, but the command is something like this (on macos anyway):
1: Tune to the POGSAC frequency in your SDR software of choice.
2: Send the NFM modulated audio to a virtual sound card.
3: Use the `sox` program to convert the audio from the virtual sound card and pipe to multimon-ng.

```
sox --no-show-progress -t coreaudio "BlackHole 16ch" --channels 16 --bits 32 --encoding floating-point --rate 48000 -t raw --channels 1 --bits 16 --encoding signed-integer --rate 22050 - | ./multimon-ng --timestamp -c -a POCSAG512 -a POCSAG1200 -a POCSAG2400 -a FLEX -a SCOPE -
```

Of course, if you're on Linux, you'll need to replace coreaudio with whatever sound driver you're using.

jules35, 2023-02-15:
Hi

thanks for your answer

ok for this part, it's much more complicated than with rtl_fm thet you can pipe directly to multimon-ng.
what command line sdr software can i use with the rf hack?
is there an equivalent of rtl_fm like rx_fm for example?
is the hackrf_transfer commande with -w can pipe directly to multimon?

THANKS
Julian

dmaltsiniotis, 2023-02-15:
There is no command that I know of that you can use in the same way as rtl_fm, but that doesn't mean they don't exist since my research was done a few years ago and that time I did not find any.

hackrf_transfer works at a much lower level with raw IQ radio data, you cannot use it to pipe directly into multimon. What you'd need to do first is demodulate the signal, then pipe that audio into multimon. There are a number of tools that might be able to do a basic NFM demodulation from the command line but you'll have to do some research first.

miek, 2023-02-15:
There's [rx_tools](https://github.com/rxseger/rx_tools) which implements rx_* equivalents of the rtl_* commands, for any devices that are supported by SoapySDR. It hasn't been updated in a while, but is probably worth a try.

dmaltsiniotis, 2023-02-16:
Using @miek's suggestion, I was able to compile rx_tools locally and after some playing around with the parameters (which some don't fully make sense and some outright don't work) I was able to get multimon-ng to decode POCSAG with this command:

```
# Listen to validate the signal
./rx_tools/rx_fm -d device=hackrf -f 929.622M -s 22050 | play -r 22050 -t raw -e s -b 16 -c 1 -V1 -

# Decode with multimon-ng
./rx_tools/rx_fm -d device=hackrf -f 929.622M -s 22050 | ./multimon-ng/build/multimon-ng --timestamp -c -a POCSAG512 -a POCSAG1200 -a POCSAG2400 -a FLEX -
```

Of course, replace the command line paths with your tool paths. @jules35  I think this about as simple as a command line is possible in this instance.

Side-note: I suspect the sample rate defaults are not set to an optimum value for a HackRF One, but changing them anything else such as 8,000,000 for optimal ADC performance, yielded an output sample rate that was unusable without an intermediary conversion with `sox`. The -r resample flag seems to not work correctly, but even the rtl_fm docs say not to use it and to use sox instead for resampling.

closed 2023-03-18 by straithe