greatscottgadgets/hackrf #1286: Alternative Windows host tools binary download source

dmaltsiniotis, 2023-02-18:
### What feature would you like to see and why?

It looks like the Pothos SDR folks have stopped updating their packages sometime in July 2021, which traditionally have included the HackRF host tools for Windows, and are referenced in the Windows setup docs [here](https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html#windows-binaries).

We may want to consider either
a) Finding an alternative source for Windows users to download the latest Windows host tools binaries or
b) Adding the binaries directly along side the releases here in GitHub.

There was some talk a while back (perhaps in the form of a pull request I can't remember exactly) to make the artifacts generated by the GitHub Actions pipeline available for download (for example windows binaries of the host tools.) This would solve the problem as the the build pipelines are already doing the hard work of creating the Windows binaries, we would just need to add some plumbing to be able to make them accessible to users.

straithe, 2023-02-18:
Thank you for the suggestion!

martinling, 2023-02-18:
Yes, we need an alternative solution for Windows binaries if PothosSDR isn't going to continue - this is something I've raised internally as well.

What makes this difficult is that just having a build of libhackrf and the host tools on its own is not very helpful. You also need the rest of the software available - GNU Radio, Soapy/Osmo, and all the applications. And it all needs to be built using the right version of MSVC, and installed such that it uses the right version of libhackrf. Windows doesn't have a solution for managing binary dependencies in separate packages, so it all has to be bundled together.

PothosSDR has been exactly what's needed, and I think the best solution would be for that project to be continued in some form. There is an [open issue](https://github.com/pothosware/PothosSDR/issues/97) from November asking about future updates, but no response so far.

The build scripts are all in the PothosSDR repository, so in theory there is nothing stopping someone from forking it, updating some of the packages it uses and setting up automated builds using GitHub Actions. But maintaining the whole project going forward is a big job.

dmaltsiniotis, 2023-02-18:
I think there's still a lot of value in only providing the host tools and library though. Take for example the new user experience on Windows: Let's I just bought a HackRF One, and I'd like to "install" it and start scanning. I don't need a GNU Radio / Soapy or anything outside of the host tools for maybe updating firmware and/or using newer features of hackrf_sweep. Additionally hackrf_info for debugging/troubleshooting. I could be wrong on this point but I thought most of the major SDR scanners come bundled with their own Soapy implementations (SDR Sharp, SDR++, CubicSDR, etc...) so there's no need for any of the ancillary tooling provide by Pothos.

It would also help have links to direct Windows users (who commonly ask in the Discord) to the latest host tools when there are firmware updates, so they get the latest version of the host tools as well.

martinling, 2023-02-18:
I agree that it would be useful for us to have a download available which includes binaries of libhackrf and the host tools.

From a support perspective, I think we'd want that to be an MSI installer which takes care of installing everything to a standard location, replacing any existing version there, and adding the tools to the `PATH`. If it's just a zip file with a DLL and some .exe files it's going to cause a lot of confusion and questions.

keukenrol, 2023-12-11:
I would like to see the windows binaries being available stand-alone too. I've had multiple compatibility issues with the PothosSDR install combined with radioconda.

dmaltsiniotis, 2023-12-11:
@keukenrol I have a branch that is [90% there](https://github.com/dmaltsiniotis/hackrf/tree/feature/package-windows-artifacts) (still fighting with some esoteric CMAKE commands) for building and packaging the windows binaries as part of the regular build cycle. It's not ready for PR yet, but you can find the current binaries on my branch here for the time being: https://github.com/dmaltsiniotis/hackrf/releases/tag/v2023.01.1

dmaltsiniotis, 2023-12-29:
Any CMAKE gurus out there to help me with the last 10% of this PR? https://github.com/greatscottgadgets/hackrf/pull/1391

The issue is that the cmake install step is inconsistent between my local environment and the build server. For some reason, I cannot get the cmake install step to correctly copy the 3rd party lib's for libhackrf and hackrf-tools to the main install folder (which is then packaged (zipped) up and uploaded as a build artifact. Once this last hurdle is resolved this PR will fix add a packaged windows zip to every nightly/weekly build.

closed 2024-03-07 by mossmann