greatscottgadgets/hackrf #1289: Unrecognized Board ID

Zero-Cool-1, 2023-02-22:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

To output a signal with the HackRF

### What outcome actually happened?

I'm unable to output my RF source and it cannot be recognized by GNU.

### What operating systems are you seeing the problem on?

Rasbian 10, 11/Ubuntu 22.04

### What is the output of ```hackrf_info```?

nasa@pi-sdr:~ $ hackrf_info
hackrf_info version: git-6e5cbda
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000010961dc2a3b184f
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00784f6d
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff

### Are you using any third-party software?

GNU 3.7.14.0

### Are you using any third-party hardware?

No

straithe, 2023-02-22:
Can you please add a picture of your HackRF?

Zero-Cool-1, 2023-02-22:
[image]

straithe, 2023-02-22:
Thank you. Can you please upload a picture of your flow graph?

martinling, 2023-02-22:
You have a HackRF One revision 9 which is the latest version. Because it has some different components, it has a new board ID, so your older version of `hackrf_info` does not recognise it. If you update your libhackrf and host tools to the 2023.01.1 release, it will be correctly identified by `hackrf_info`.

Aside from what is reported by `hackrf_info` and flashing firmware with `hackrf_spiflash`, the difference should not affect your ability to use any software that supports the HackRF One. What problems are you seeing with using it?

Zero-Cool-1, 2023-02-22:
[image]

Zero-Cool-1, 2023-02-22:
I guess we overlooked that when performing apt-get install libhackrf... it only installs version v2018.01.1-2.

We're unable to perform transmitting because GNU sees it as "not connected," which I thought was because of the missing board ID.

straithe, 2023-02-28:
@Zero-Cool-1 has updating libhackrf and your host tools addressed the problem you were experiencing?

JohnCanty, 2023-03-27:
This bug description is pretty excellent. So thank you to the OP of this bug. I have a similar issue. I haven't updated libhackrf yet. I will. Just a side note. I am noticing that this is preventing a portapack H2 from booting. At least I think this is the problem. When I flash the mayhem firmware it pseudo bricks the board, as I have to DFU recover the board. Any thoughts on if this could be the issue? I know you don't maintain the mayhem firmware, but I figured I would add some of this to the bug if someone else is out looking for it.

straithe, 2023-04-02:
I'm closing this as there hasn't been a response from the original poster in over 30 days. Please re-open this issue or open a new one if you still need assistance.

straithe, 2023-04-02:
@JohnCanty Some work has bee done to get Portapack working with the latest HackRF revision. Please update to the latest Mayhem code. Hopefully that will help .

closed 2023-04-02 by straithe