greatscottgadgets/hackrf #1293: Which driver to use on Windows?

sa7mon, 2023-03-02:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

`hackrf_info.exe` to work after firmware upgrade

### What outcome actually happened?

I'm getting a little frustrated - I can't seem to even tune to FM stations with my HackRF One so I thought I should start at the beginning of troubleshooting and make sure the firmware is up-to-date. Here's how that went:

```
PS C:\Program Files\PothosSDR\bin> .\hackrf_info.exe --version
libhackrf version: git-7047252 (0.6)
Found HackRF
Serial number: 0000000000000000f77c60dc233aa8c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
```

```
PS C:\Program Files\PothosSDR\bin> .\hackrf_spiflash.exe --write C:\Users\dan\Downloads\hackrf-2023.01.1\firmware-bin\hackrf_one_usb.bin
File size 42500 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 42500 bytes at 0x000000.
```

(Unplugged & plugged back in)

```
PS C:\Program Files\PothosSDR\bin> .\hackrf_info.exe --version
hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
```

I suspect maybe I'm using the wrong USB driver? Here are the options I see in Zadig 2.8:

-    WinUSB (v6.1.7600.16385)
-    libusb-win32 (v1.2.7.3)
-    libusbK (v3.1.0.0)
-    USB Serial (CDC)

Additionally, I can either "Install Driver" or "Install WCID Driver"

### What operating systems are you seeing the problem on?

Windows 10 x64

### What is the output of ```hackrf_info```?

```
PS C:\Program Files\PothosSDR\bin> .\hackrf_info.exe --version
hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
```

### Are you using any third-party software?

PothosSDR for hackrf and zadig binaries so I don't have to compile them.

### Are you using any third-party hardware?

No, just the stock extendable antenna

martinling, 2023-03-02:
You shouldn't need to use Zadig at all. The HackRF One has a WCID descriptor which tells Windows which driver to use for it (WinUSB). That driver should be assigned automatically. Use the "Install WCID Driver" option to set it back to that default.

sa7mon, 2023-03-02:
Whew, okay I uninstalled the device in Device Manager and now `hackrf_info` reports the following

```PS C:\Program Files\PothosSDR\bin> .\hackrf_info.exe
hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc233aa8c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00614f6e
```

I think that looks right? My next question is - is there an easy way (probably without SDR#) to verify that my device is working? I found a simple fm_radio python script but I guess it's really outdated and the package names changed. 

I would just like a simple tool (minimum requirements) that will let me tune to a local FM station to verify all is good

martinling, 2023-03-02:
Personally I think using one of the graphical receiver programs (SDR#, SDRConsole, SDR++, gqrx, etc) is actually best, because then you can see the spectrum, see when and where there are signals to listen to, and see how the reception responds to changing the gain settings. With a command line tool you have to get all the settings right to get anything.

sa7mon, 2023-03-10:
I was able to get it working! I'm not sure what settings I had messed up in SDR#, but I loaded up an older version I also had installed which apparently had my old settings saved and I could tune FM stations just fine.

Thanks for the help!

closed 2023-03-10 by sa7mon