greatscottgadgets/hackrf #1294: No Signal from HackRF One Showing on SDR# - Flatline Noise Only

Culdee, 2023-03-13:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

On SDR# I'm expecting to see various FM broadcasts between 88 and 108 MHz, but I see nothing except flatline signal noise. This is my first time using this HackRF One device after returning from a 1 year deployment. It could have been dead-on-arrival but I never had the chance to check.

[image]. 

### What outcome actually happened?

[image]

### What operating systems are you seeing the problem on?

Windows 10 Pro w/ most recent updates

### What is the output of ```hackrf_info```?

'hackrf_info' is not recognized as an internal or external command,
operable program or batch file.

### Are you using any third-party software?

I'm using SDR#

### Are you using any third-party hardware?

N/A

miek, 2023-03-13:
Have you increased the LNA & VGA gain sliders? (they can be accessed from the settings cog icon top left) If I remember correctly, SDR# has the gains set to minimum by default.

Culdee, 2023-03-13:
I just tried adjusting the LNA and VGA sliders to an uncomfortably high level. No meaningful difference.

[image]

Culdee, 2023-03-13:
I wonder why the signals are so weak?  I can get better reception with a paperclip than the included HackRF One attenna.  In fact there's little difference between leaving the antenna completely off and having it in now that I've tried it.

Is the included antenna really that bad it can barely see an FM signal?

miek, 2023-03-13:
Could you post a photo of your HackRF & antenna?

Culdee, 2023-03-13:
Yes, here is. Thanks for the help by the way!

[image]

miek, 2023-03-13:
> I wonder why the signals are so weak? I can get better reception with a paperclip than the included HackRF One attenna. In fact there's little difference between leaving the antenna completely off [...]

No, the ANT500 antenna should definitely perform better than that, so it sounds like it is defective. If you get in touch with the distributor you purchased it from, they will replace it for you.

Culdee, 2023-03-14:
Are you saying the antenna is defective or the HackRF One?

Culdee, 2023-03-14:
Is there any way I can know that it isn't also the HackRF that might be having an issue?

miek, 2023-03-14:
From the screenshot you posted above and the test you did where a paperclip improved the reception I'm confident that the HackRF is fine and it's just a faulty antenna.

Also, I meant to add before, but you don't need to be concerned about setting the LNA/VGA gains high - it's normal to have those set around 20 to 30dB each for general use.

martinling, 2023-03-14:
There has to be a connection internally from the centre pin of the antenna connector through to the extendable part of the antenna. And those parts also need to be isolated from the outer rim of the antenna connector, which connects to ground on the HackRF side. If there's no connection from the pin to the body of the antenna, or if that path is shorted to ground, then the antenna can't work.

If you're curious, you can probe those parts with a meter to see what's gone wrong, but it's clear enough from comparing to a paperclip test that there's a fault.

Culdee, 2023-03-16:
I see. I have a multimeter. How do I test for a short using the two probes of the multimeter?

martinling, 2023-03-18:
You would measure the resistance from the centre pin of the connector to the metal part of the antenna, and from either of those places to the outer rim of the connector. A short would show up as a resistance close to zero.

closed 2023-05-04 by straithe