greatscottgadgets/hackrf #1297: Hardware version r9 not detecting external clock

weikinhuang, 2023-03-16:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Hi I have recently bought a genuine version of the HackRF One. This one is a `r9` hardware. I'm trying to install a external TCXO in the P22 headers. Looking at the docs for the changeset for r9, it says the Si5351 was switched from C to A, but the block diagram says it can still accept a 10MHz clock.

I've also checked the other threads regarding the clock not being detected, but they all predate the r9 hardware release.

Thanks!

### What are the steps to reproduce this?

I've tried installing the [Nooelec Tiny TCXO](https://www.nooelec.com/store/sdr/sdr-addons/tiny-tcxo.html). And running `hackrf_debug --si5351c -n 0 -r` but no matter what I get back `[  0] -> 0x51`. Does r9 of the hardware not support an external TCXO?

### Can you provide any logs? (output, errors, etc.)

```
hackrf_info --verbose
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000010...........
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006a4f4e
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```

martinling, 2023-03-16:
The r9 hardware supports an external clock on either the CLKIN port or the P22 header, as per earlier revisions.

However `hackrf_debug --si5351c -n 0 -r` is not expected to return the same values, since r9 doesn't use the Si5351C.

The `hackrf_clock -i` command should return correct results, though. Perhaps your TCXO is not working? Do you have any means of checking the clock signal, e.g. with an oscilloscope?

weikinhuang, 2023-03-16:
I have 2 of those modules and they both report the same with hackrf_clock -i. I do have an oscilloscope, which pin should I probe?

martinling, 2023-03-16:
CLKIN is on pin 2 of P22, This pin is also directly connected to the centre conductor of the CLKIN SMA port.

weikinhuang, 2023-03-16:
My oscilloscope is reporting no signals, also checked while running spectrum analyzer in URH.

martinling, 2023-03-16:
That does sound like the TCXO isn't working, but it's strange that you have the same result with two of them.

It shouldn't matter whether the HackRF is receiving or not. The TinyTCXO is always on - it gets its power supply from VCC (3.3V) on pin 11 and GND on pin 10. As long as the HackRF is powered you should see its clock signal on pin 2 and also on the CLKIN SMA port.

weikinhuang, 2023-03-17:
Ok, actually, my oscilloscope is reporting a 10MHz waveform when plugged into the hackrf's clkin sma center pin with the TCXO plugged in. However `hackrf_clock -i` is still reporting `CLKIN status: no clock signal detected`. I also tested the TCXO by itself with a 3v3 source, and that's also reporting 10MHz.

mossmann, 2023-03-17:
Can you probe the highlighted end of R39 with your oscilloscope?
[image]

weikinhuang, 2023-03-17:
no signal, flat response on the scope

weikinhuang, 2023-03-17:
R165 is not populated, and the 10MHz signal makes it to the R165-1 pad. R38 is 0V on either side.

mossmann, 2023-03-17:
The signal isn't getting through U26 for some reason, perhaps because it is not soldered correctly. This is something that was tested at the factory, but maybe there was a subtle soldering failure that didn't come loose until shipping. Please seek a replacement from your reseller. (If you are handy with hot air, feel free to see if reworking U26 fixes the problem. If it doesn't work, we'll still authorize a replacement.)

weikinhuang, 2023-03-17:
I actually have a second one on hand, due to me asking for a replacement because I thought it was defective, let me try the other one @mossmann.

mossmann, 2023-03-17:
Actually, before you modify anything, what is the voltage of the 10 MHz signal from your TCXO?

weikinhuang, 2023-03-17:
@mossmann just tried the other one, it has the exact same issue, the clock is available at R165, and is flat at R38. The clock waveform is ~100mV at CLKIN and ~200mv at R168 pad 1.

mossmann, 2023-03-17:
That isn't enough voltage swing. Can you measure the voltage of the unloaded TCXO output (powered but not connected to the HackRF's CLKIN)?

weikinhuang, 2023-03-17:
sure, at 3V3 input, it still generates a 100mV p-p signal. That's really strange because according to nooelec's site, this was designed for the Hackrf.

weikinhuang, 2023-03-17:
Whoops, My probe was at 10X, I switched it to 1X, and the p-p is actually 1V

weikinhuang, 2023-03-17:
When loaded on the hackrf, the p-p is ~500mV at 1x

mossmann, 2023-03-17:
Even at 1Vp-p that isn't enough voltage. We've always specified that the clock input signal should be a 3.3V square wave: https://hackrf.readthedocs.io/en/latest/external_clock_interface.html

There is something wrong with the TCXO. It is possible that it is a design flaw but that they got lucky and it worked anyway on older HackRF One hardware revisions.

One thing you can do to confirm that the HackRF's clock input is working is to connect it to the other HackRF's clock output. You can enable the output clock with `hackrf_clock -d <serial_number1> -o 1` and then check the other HackRF's input with `hackrf_clock -d <serial_number2> -i`.

weikinhuang, 2023-03-17:
Alright, connecting the 2 Hackrf from CLKOUT to CLKIN on the second did work
```
$ hackrf_clock -i
CLKIN status: clock signal detected
```

mossmann, 2023-03-17:
I haven't used a made-for-HackRF TCXO module myself. I *would* have recommended the one from NooElec, but it seems there is a problem there. We'll contact NooElec about this.

A quick search reveals this design which looks good to me, but I'm not sure if anyone is selling it (exactly as designed): https://github.com/osqzss/gps-sdr-sim/tree/master/extclk

weikinhuang, 2023-03-17:
Awesome, thanks @mossmann!

uulltt, 2023-09-07:
Sorry if this is a newbie question, but when I try running `hackrf_clock -i`, I get `hackrf_clock: invalid option -- 'i'` and it just prints the help menu. 
I have the Tiny TXCO installed because when I run `hackrf_debug --si5351c -n 0 -r` I get `[  0] -> 0x01`. 
Running `hackrf_info` gives the following:
```
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000087.........
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x004c4f5e
```
Maybe there's just something I forgot to install but if anyone can help out I would greatly appreciate it.

martinling, 2023-09-07:
@uulltt You have an older version of libhackrf and the host tools.

The `-i` option was added in the 2022.09.1 release as a simpler, hardware-independent way to check the CLKIN status.

uulltt, 2023-09-07:
> @uulltt You have an older version of libhackrf and the host tools.
> 
> The `-i` option was added in the 2022.09.1 release as a simpler, hardware-independent way to check the CLKIN status.

Do I have to build from source in order to update it? I initially just installed whatever was already on APT.

martinling, 2023-09-07:
You can get a newer version via apt if you use the right repositories. Debian has 2022.09.1 in `stable` and 2023.01.1 in `testing`.