greatscottgadgets/hackrf #1298: how to upgrade libhackrf

garryharr, 2023-03-24:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

i want my hackrf host libhackrf to the latest version

### What outcome actually happened?

hey i need to upgrade my hackrf to the 2023 version in raspberry4 and download the new release, flash the firmware. in rasp i follow the instruction in the help page and compile the host directory. but i dont know how to "upgrade" libhackrf and it always shows an unknown version 0.5. no surprise when i plug in hackrf there are errors such as undefined symbols.

### What operating systems are you seeing the problem on?

raspberry pi 4b

### What is the output of ```hackrf_info```?

[image]

### Are you using any third-party software?

no

### Are you using any third-party hardware?

no

martinling, 2023-03-24:
If you built the whole `host` directory, rather than just `hackrf-tools`, then `make install` should have already installed the library in `/usr/local/lib`. You may need to run `sudo ldconfig` to update the library cache. If that doesn't pick up the new version, check in `/etc/ld.so.conf` whether `/usr/local/lib` is included in the library search path.

garryharr, 2023-03-25:
thanks a lot, do i have to build hackrf-tools and libhackrf first and then build the host directory

garryharr, 2023-03-25:
just solved. remove the old libhackrf package and things work well. thanks

closed 2023-03-25 by straithe