greatscottgadgets/hackrf #1303: External Crystal

apppie123, 2023-04-07:
### What would you like to know?

I've been having issues with my HackRF device with external 0.1ppm TCXO from amazon.
The HackRF used to work fine to transmit GPS signals, but since a week the GPS receiver is starting to have issues with getting a lock. I have the HackRF powered 24/7 for a month or two now.

This is the crystal I'm using:
https://www.amazon.com/Precision-External-PPM0-1-HackRF-Application/dp/B07NXSFFLB/ref=sr_1_2?crid=1Y91GHLNV7A0V&keywords=TCXO&qid=1680886960&sprefix=tcxo%2Caps%2C158&sr=8-2

Perhaps this one is prone to drift over time? Current temperature in the room is about 10 degree celsius.

I ordered the 0.5ppm from NooElec but I believe that one doesn't work with the more recent HackRF's.
I have not found any other crystals, other than ordering PCB's myself and soldering one together.

Anyone have any ideas? Perhaps suggestions for an OCXO?
https://www.amazon.com/Frequency-Standard-Oscillator-Reference-Constant/dp/B0BQYXYKHJ/ref=sr_1_40?crid=XK3Q362ROW1U&keywords=OCXO&qid=1680887152&sprefix=ocxo%2Caps%2C142&sr=8-40

straithe, 2023-04-07:
Please know that some TCXOs do not meet the 3.3 V signal requirement of HackRF One even though they are advertised as working with HackRF. These TCXOs happened to work with older hardware versions of HackRF One, but do not work with the latest revision. I suggest double checking any TCXOs you are looking at to make sure they meet that 3.3 V requirement.

apppie123, 2023-04-07:
I have read that in another issue(I think I went through all of them by now). This first one has been working for weeks without issues, so I don't know if the TCXO is the issue(due to temperature?) or if there is something else. I don't have a freq. counter unfortunately.

It's hard to find a compatible TCXO or OCXO that would not require to order circuitboards.

straithe, 2023-05-04:
Your best way to move forward is to contact the company you purchased your TCXO from.

closed 2023-05-04 by straithe