greatscottgadgets/hackrf #1306: When installing grOsmosSDR, hackrf isn't enabled

Hendrik-Valcke, 2023-04-10:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I'm trying to install grOsmosSdr to work with the hackrf in GNU Radio. I'm following the instructions from https://osmocom.org/projects/gr-osmosdr/wiki/GrOsmoSDR  (I installed hackrf beforehand)
I was hoping to get something like it says on the website:
- ######################################################
-- # gr-osmosdr enabled components                         
-- ######################################################
--   * Python support
--   * Osmocom IQ Imbalance Correction
--   * sysmocom OsmoSDR
--   * FUNcube Dongle
--   * FUNcube Dongle Pro+
--   * IQ File Source
--   * Osmocom RTLSDR
--   * RTLSDR TCP Client
--   * Ettus USRP Devices
--   * Osmocom MiriSDR
--   * HackRF Jawbreaker
--   * nuand bladeRF
--   * RFSPACE Receivers
-- 
-- ######################################################
-- # gr-osmosdr disabled components                        
-- ######################################################
-- 
-- Building for version: v0.1.0-4-g37aba331 / 0.1.1git
-- Using install prefix: /usr/local

### What outcome actually happened?

Instead I only get a couple of enabled components:
######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * Python support
--   * IQ File Source & Sink
--   * RTLSDR TCP Client
--   * RFSPACE Receivers
--   * Red Pitaya SDR
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * Osmocom IQ Imbalance Correction
--   * FUNcube Dongle
--   * Osmocom RTLSDR
--   * Ettus USRP Devices
--   * Osmocom MiriSDR
--   * HackRF & rad1o Badge
--   * nuand bladeRF
--   * AIRSPY Receiver
--   * AIRSPY HF+ Receiver
--   * SoapySDR support
--   * FreeSRP support
--   * XTRX SDR
-- 
-- Building for version: 0.2.0.0 / 0.2.0
-- Using install prefix: /usr/local

### What operating systems are you seeing the problem on?

Ubuntu

### What is the output of ```hackrf_info```?

hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2fa861e1
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00704762

### Are you using any third-party software?

GNU Radio

### Are you using any third-party hardware?

N/A

miek, 2023-04-10:
How did you install the HackRF software? If it was using the Ubuntu packages, make sure to also install the package `libhackrf-dev` and then try the gr-osmosdr build again.

Hendrik-Valcke, 2023-04-10:
> How did you install the HackRF software? If it was using the Ubuntu packages, make sure to also install the package `libhackrf-dev` and then try the gr-osmosdr build again.

I believe it was indeed just through ubuntu packages, I'll try installing the libhackrf.

Hendrik-Valcke, 2023-04-10:
> > How did you install the HackRF software? If it was using the Ubuntu packages, make sure to also install the package `libhackrf-dev` and then try the gr-osmosdr build again.
> 
> I believe it was indeed just through ubuntu packages, I'll try installing the libhackrf.

Didn't seem to fix it, I also have a broken package that I'm not able to fix:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 soapysdr-module-lms7 : Depends: soapysdr0.7-module-lms7 but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

```
When I run the apt --fix broken install command it ends with a dpkg error:
```
Unpacking soapysdr0.7-module-lms7:amd64 (22.09.1-myriadrf1~ubuntu22.04) ...
dpkg: error processing archive /var/cache/apt/archives/soapysdr0.7-module-lms7_2
2.09.1-myriadrf1~ubuntu22.04_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.8/libLMS7Suppo
rt.so', which is also in package soapysdr0.8-module-lms7:amd64 20.10.0+dfsg-5
Errors were encountered while processing:
 /var/cache/apt/archives/soapysdr0.7-module-lms7_22.09.1-myriadrf1~ubuntu22.04_a
md64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

Hendrik-Valcke, 2023-04-13:
I retried installing all the packages (gnuradio, hackrf, osmosdr) from scatch using the ubuntu apt-get install command and it all seems to work now

closed 2023-04-13 by Hendrik-Valcke