greatscottgadgets/hackrf #1309: Command line tool to switch antenna power on/off, similar to rtl_biast

ai6aj, 2023-04-19:
### What feature would you like to see and why?

Some software (notably SDR#) does not provide a way to turn on antenna power to power an offboard LNA.  On RTL-SDR the easy workaround is to use rtl_biast to enable the power circuit, then start your software:

$ rtl_biast -b 1

I cannot find a method of doing this for HackRF.  I have written hackrf_biast.c and placed it in my fork but won't submit a pull request if there's an official (or just better) way of doing this.

nmoon88, 2023-04-23:
I requested exactly the same thing last night. It seems like this is something that would benefit everyone but the developers don't want to do it. I was told to hire a C developer to code such a utility in the Discord chat.

martinling, 2023-04-23:
This can't be done purely with a host-side utility, it would also need firmware changes. Currently the firmware turns off the bias tee every time it leaves TX or RX mode.

ai6aj, 2023-04-24:
> This can't be done purely with a host-side utility, it would also need firmware changes. Currently the firmware turns off the bias tee every time it leaves TX or RX mode.

As long as the power stays on to _start_ receiving, that's good enough.  I have written the utility and submitted a pull request, antenna power stays on in SDR# until you hit STOP or exit the program... at which point you need to re-enable it again.  But that's a lot better than not being able to turn it on at all.

martinling, 2023-04-25:
I had a quick chat with @mossmann about this, and we think a useful direction for development would be to add a bias tee control mode setting to the firmware, which would have multiple options for how the bias tee would be controlled:

- Host controlled (default, current behaviour).
- Turn on automatically during RX only.
- Turn on automatically during TX only.
- Turn on automatically during both TX and RX.
- Turn on and leave on.
- Turn off and leave off.

This would provide better control over how the bias tee should be used in software that doesn't control it directly. It would also be useful even for software that's aware of the bias tee, since it would be possible to speed up and simplify mode changes by telling the firmware to control the bias tee automatically, rather than having to request changes each time.

There would need to be a new USB vendor request type and a new libhackrf function to set the bias tee control mode. Then a command line tool like @ai6aj's `hackrf_biast` proposal could be written to control it.

closed 2023-09-13 by mossmann