greatscottgadgets/hackrf #1310: My HackRF is receiving a Signal at least 20db less than it Should

erd0spy, 2023-04-22:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

## Steps to reproduce
1. Used 2 HackRF One devices with the same antenna, same software, same USB port, and at the same time
2. Utilized GQRX (latest version)
3. Conducted an experiment: Attempted to receive any signal

### What outcome actually happened?

I had borrowed a HackRF from my Professor because the signal strength of my HackRF (for both RX and TX) was very low and could not receive almost any signal except FM signals. Although both appeared to be the same, in my measurements with GQRX and other software, I noticed that my own HackRF received at least 20dB less signal strength than my Professor's HackRF. In my tests with a signal analyzer in the laboratory, I could hardly see the signal I was sending. This was an issue I had not encountered with other HackRFs.

HackRF Images:

The one belonging to my professor was on the left, and mine was on the right

[image]

I tested with the same antenna, same USB cable, same USB port, and at the same time.

## GQRX Images:

My Professor's HackRF:

[image]

My HackRF:

[image]

## PCB of My HackRF:

Front:
[image]

Back:
[image]

### What operating systems are you seeing the problem on?

Ubuntu 20.04

### What is the output of ```hackrf_info```?

```bash
$ hackrf_info
hackrf_info version: git-ebe1ca00
libhackrf version: git-ebe1ca00 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82547b45f
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: 0xa000cb3c 0x0057435c
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
```

### Are you using any third-party software?

Gnu Radio, GQRX

### Are you using any third-party hardware?

No

martinling, 2023-04-22:
The screenshots don't show your gain settings. Were you using the same settings on the "Input controls" tab in both cases? I think GQRX may save these separately for different units.

erd0spy, 2023-04-22:
> 

Yes, I am using same gain settings.

erd0spy, 2023-04-22:
> The screenshots don't show your gain settings. Were you using the same settings on the "Input controls" tab in both cases? I think GQRX may save these separately for different units.

Yes, I am using same gain settings.

erd0spy, 2023-04-22:
When I transmitted a signal using hackrf_transfer with my own HackRF, I was unable to see any results on the signal analyzer.

erd0spy, 2023-04-22:
I'm certain that something is malfunctioning for both TX and RX, even though the firmware and device seem to be functioning properly.

martinling, 2023-04-22:
I noticed from your photos that the CLKOUT port is damaged, as if perhaps the unit has been dropped and landed on it.

If the board has been exposed to a large shock then there are a lot of possible ways it may have been damaged, e.g. cracked capacitors or solder joints, which might explain the performance issues.

Are you testing with or without the RF amplifiers enabled? Those are fairly easy to replace, but damage elsewhere may be harder to track down.

erd0spy, 2023-04-23:
LNA and VGA Gain seem to work. Although it does not allow me to receive more signals, it increases the value of the current signal. When I test it in C programs, the value HACKRF_SUCCESS is returned for the lna_gain, vga_gain, amp_enable, and antenna_enable functions (although I'm not sure this can be used for hardware failure testing). Unfortunately, it may have been damaged by hardware damage.

straithe, 2023-05-26:
I am closing this issue as hardware damage, as suggested by @martinling, is likely the cause of the issues you are seeing.

closed 2023-05-26 by straithe