greatscottgadgets/hackrf #1311: stand alone Bias-T control utility

nmoon88, 2023-04-23:
### What feature would you like to see and why?

I really would like an external utility to enable and disable bias-t for the HackRF One and similar devices. Being forced to rely on certain programs to incorporate it severely limits what can be done and such a utility already exists for RTL-SDR sticks. This would be useful on Windows, Linux and MacOS or any other platform that supports the device.

ai6aj, 2023-04-24:
I've submitted a pull request to add this to the main repo but if you need the utility ASAP you can find it in my fork: https://github.com/ai6aj/hackrf/tree/20230418_suitejo_biast

One caveat, the HackRF firmware turns off the bias T power whenever it leaves RX mode.  So once you hit STOP or exit SDR# (or another program) the external power will shut off and you'll need to re-enable it.  IMO it's still much less of a PITA than using external power for your LNA or other device.

martinling, 2023-04-25:
We have two issues open for the same thing here. Let's continue the discussion on #1309. I'll close this one.

closed 2023-04-25 by martinling