greatscottgadgets/hackrf #1315: Is it possible to have an input reference signal on one of the Opera Cake ports with it in time-mode for PDDF?

Nistp, 2023-04-28:
### What would you like to know?

As my title states, is it possible to put a reference signal on one of the ports of the Opera Cake board or the HackRF itself with the Opera Cake board attached in time mode with an antenna array? If so, would the reference signal be put on B0, for example, if my antenna array is on A1-A4? 

How would the Opera Cake board be configured then? Manually turn on the B0 port and then do a time mode for the antenna array? Attach the reference to a HackRF port?

If this is not possible out of the box, which source files should I look to attempt to change the firmware to do what I'd like to do?

Thank you!

martinling, 2023-04-29:
There's only one antenna input on the HackRF, so where do you want the reference signal to end up?

If you just want to include it in the switching sequence along with an antenna array, you can put both the array and the reference on any of the secondary ports (A1-A4 & B1-B4).

E.g. put your antenna array on A1-A4, the reference on B1, and use:

`hackrf_operacake --mode time -w 1000 -t A1 -t A2 -t A3 -t A4 -t B1`

Or if you want to switch back to the reference in between sampling each antenna:

`hackrf_operacake --mode time -w 1000 -t A1 -t B1 -t A2 -t B1 -t A3 -t B1 -t A4 -t B1`

Nistp, 2023-04-30:
@martinling  thank you for your answer! Ideally, what I'd like to do is to be able to have the reference signal being constantly read and have a comparison with the signals coming in from the array to determine the phase shift. In GNU Radio I'd say that'd be equivalent to me having two sources - one for the reference signal and one for the cycling array. 

In other words, one of the ports would be doing it's own thing and the other secondary ports will be doing their own thing. Is my description making sense/helping?

martinling, 2023-04-30:
To do that you need an SDR with two RX inputs that it can sample simultaneously. HackRF only has one. Opera Cake lets you switch between multiple sources, but you can still only be capturing one of them at a time.

If you're switching quickly enough though, you can capture enough of the waveform on each input to be able to measure the phase shifts between them. Using the [pseudo-doppler](https://greatscottgadgets.com/2022/09-19-pseudo-doppler-redux-shmoocon-2018/) technique this can be used to implement direction finding, even without a reference input.

Nistp, 2023-04-30:
@martinling and the HackRF + OperaCake does not have two RX inputs, the antenna switch simply let's you switch between the secondary ports, but none of them are mixed together? In other words, this is a hardware limitation, not a firmware one, correct?

Is there a way to make that work with 2 HackRFs (1 with OperaCake and 1 without) and a synchronized clock source?

And onto the last one - what is the max possible switching speed? 20 MSPS - 20 MHz?

martinling, 2023-04-30:
> the HackRF + OperaCake does not have two RX inputs, the antenna switch simply let's you switch between the secondary ports, but none of them are mixed together? In other words, this is a hardware limitation, not a firmware one, correct?

Correct, but note that you can switch them very quickly. And from a mathematical point of view, switching rapidly between two inputs at a fixed frequency is the same as multiplying each input by a square wave of that frequency. So you can think of the switch as a kind of mixer, in which the switching rate is the LO frequency.

Consider also the case in which you sample at say 2MHz, and switch inputs every sample. If you then split the data into odd and even sample numbers, you effectively have near-simultaneous measurements of both inputs, taken at 1MHz sample rate each.

These sorts of mathematical relationships make various sorts of multi-channel signal analysis possible, even though there is only one RX input on the HackRF.

> Is there a way to make that work with 2 HackRFs (1 with OperaCake and 1 without) and a synchronized clock source?

Yes. You can use the [external clock interface](https://hackrf.readthedocs.io/en/latest/external_clock_interface.html) to share a clock, and [hardware triggering](https://hackrf.readthedocs.io/en/latest/hardware_triggering.html) to make them start RX at the same time.

> And onto the last one - what is the max possible switching speed? 20 MSPS - 20 MHz?

You can set the dwell time as short as a single sample, but the datasheets for the RF switches ([SKY13322](https://www.mouser.com/datasheet/2/472/201098D-13997.pdf) and [MASWSS0129](https://www.mouser.com/datasheet/2/249/MASWSS0129-9209.pdf)) specify typical switching times of 60-90 nanoseconds. So in practice at 20MSPS it would take a couple of samples for the switchover to happen.

Nistp, 2023-04-30:
@martinling thank you so much for your detailed answer, this is great, I think it'll help me try out some stuff! :)

closed 2023-05-07 by martinling