greatscottgadgets/hackrf #1317: Unknown Board ID

MarcD250, 2023-05-05:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I've recieved my HackRF One a few days ago and wanted to flash the Portapack Firmware to it. The main problem is that the HackRF is not found:
[image]

### What outcome actually happened?

I wasn't able to flash new firmware

### What operating systems are you seeing the problem on?

Linux Ubuntu

### What is the output of ```hackrf_info```?

mr@mr-CF-53SNXBYFG:~$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc2669a6c3
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00664761

### Are you using any third-party software?

No, just want to use my Portapack

### Are you using any third-party hardware?

Portapack H2, but not yet used because of the flashing issue

martinling, 2023-05-05:
Please see the guide at:

https://hackerwarehouse.tv/product-knowledgebase/portapack-h2/hackrf-one-r9-and-portapack-compatibility/

MarcD250, 2023-05-05:
I did everything according to this guide.. but without success :-(

MarcD250, 2023-05-05:
I'm Just curious because hackrf_info tells me that the Board ID is 4 (Unknown)

Board ID Number: 4 (Unknown Board ID)

Any ideas why it's Unknown?

martinling, 2023-05-05:
It's unknown because you're running an old version of libhackrf and the host tools which predates the r9 hardware. If you update to the current 2023.01.1 release, it will recognise that board ID.

MarcD250, 2023-05-06:
That was also my thought, but I tried installing new software via 'sudo apt-get install hackrf' 
But the Installer found nothing new..

straithe, 2023-05-06:
apt-get will not get you the latest versions of libhackrf and the host tools. You will have to build from the repository.

MarcD250, 2023-05-06:
Thank you! I'll give it a try.

MarcD250, 2023-05-06:
Hello, it worked! 
[image]
Thanks for all the help!

closed 2023-05-06 by MarcD250

Metalblade66, 2023-11-18:
> apt-get will not get you the latest versions of libhackrf and the host tools. You will have to build from the repository.

How can I do it?