greatscottgadgets/hackrf #1318: Amplifying hackrf transmit signal

mjcuc, 2023-05-07:
### What would you like to know?

I'm considering of testing following setup
[image]

Is there any change this setup will work?

Any risks of damaging HackRF?

straithe, 2023-05-07:
Can you please give more detail such as what you are trying to do, what results you hope to achieve, etc?

mjcuc, 2023-05-07:
Basically the following test setup on 70cm band (I have a repeater nearby):
https://www.youtube.com/watch?v=rTp6vQk5MqE

Of course, a real antenna is connected only after serious testing and with required ham licensing.

martinling, 2023-05-07:
I think this would work OK. I don't think you need any delay.

The TX/RX LEDs are switched just before the internal RF path is changed: see [here](https://github.com/greatscottgadgets/hackrf/blob/master/firmware/hackrf_usb/usb_api_transceiver.c#L318-L344) in the firmware code. So when the LED goes on/off seems like exactly the right moment to switch your external RF path as well.

Another way to control this, without needing to run extra wires to the HackRF, would be to use the built-in bias tee. This can be turned on and off by the host, and we have a PR open at the moment which enables turning it on automatically during TX (#1314). You'd need to include an RF choke and a coupling capacitor on your amp board to separate the 3.3V DC signal from the RF.

closed 2023-06-09 by straithe