greatscottgadgets/hackrf #1325: Transmit Text File using libhackrf in C/C++

PureRockets, 2023-05-16:
### What would you like to know?

All the sample code provided by ChatGPT says to use hackrf_send but this method does not exist in hackrf.h. What is the minimum C or C++ code necessary to transmit and receive a small text file? Could you provide transmit.cpp and receive.cpp?

straithe, 2023-05-16:
HackRF is open source. You can review this repository to see the code available. If you search this repository you can see that there are no transmit.cpp or receive.cpp files in this repository. I suggest reading the HackRF documentation (https://hackrf.readthedocs.io/en/latest/) to learn more about HackRF.

PureRockets, 2023-05-16:
Ok, I think I got it. At least it compiles, I'll test and follow up.
[my_hackrf.c.txt](https://github.com/greatscottgadgets/hackrf/files/11489921/my_hackrf.c.txt)

dmaltsiniotis, 2023-05-16:
@PureRockets,

Some thoughts on this: I don't think this is going to work in the way you're expecting it to. The HackRF One and its related libraries (libhackrf) have no context of protocols or file formats. They operate **purely** in the radio frequency domain, meaning that when you interact with libhackrf, it's expecting _raw radio frequency and amplitude data_ not text file bytes. You cannot arbitrarily give it any file and expect it transmit and receive successfully on other end. At best you'll get random/garbage noise transmitted, and at worst you'll end up interfering with other devices near by transmitting garbage data over them.

You were on the right initial approach: If you want to send a file and receive it using two HackRF Ones (or any SDR) you need to first come up with or select an appropriate modulation and de-modulation scheme. The modulation is what will transform the bytes of the file into radio data that libhackrf and the HackRF One can accept. GNU Radio is a tool you can use to accomplish this, as it provides the mechanisms for reading files and turning the bytes into the required raw radio data. There are quite a few examples around the internet if using GNU Radio to transmit and receive text files, and I'm sure with some research and effort you could probably reproduce a working flow graph.

Hope this helps.

Thanks,

-DM

PureRockets, 2023-05-16:
Understood, thanks for the help! I'm writing a radio service for ipfs which needs to run on a microcontroller on a CubeSat for example, and on a groundstation PC. GNU Radio has too much overhead, but I could use it to convert the file into the necessary format prior to transmitting it, for testing https://github.com/ipfs-shipyard/space/tree/main/local-dev-environment/desktop/radio-service and https://ham.stackexchange.com/questions/5450/hackrf-one-expected-hackrf-transfer-t-file-format-and-its-creation

dmaltsiniotis, 2023-05-17:
That's quite an interesting use-case, and your approach makes sense. As long as you can generate the file to send ahead of time, you can either use the pre-built hackrf_transfer tool, or code something up like your example using libhackrf directly to send it to the HackRF One. The binary file format that hackrf_transfer expects is fairly well documented/known (linked above, two signed 8-bit integers one after the other, representing the I and Q parts of a complex signal). When generating the file, I'd suggest using a sample rate somewhere in between 8,000,000 and 20,000,000 samples per second, as that's is the optimal sample rate for the ADC's and DAC's in the HackRF One.

PureRockets, 2023-05-18:
Would this Python script be sufficient to prepare a text file to quadrature format?
[ascii_to_samples.txt](https://github.com/greatscottgadgets/hackrf/files/11507669/ascii_to_samples.txt)

martinling, 2023-05-18:
You need to learn about modulation and encoding.

You have the right idea in principle, but in order to transmit this over the air and decode it on the other side, you'd need:

- The transmitter and receiver to be perfectly synchronised, such that each sample on the RX side corresponds exactly to one sample on the TX side.
- The receiver gain to be set perfectly, such that the scale of the signals at the ADC on the RX side exactly matches the scale at the DAC on the TX side.
- Zero noise of any kind to be added in the process.

None of those conditions are ever going to happen in practice, so your chance of any receiver successfully decoding this transmission are zero.

In practice, to build a working system, you need to:

- Decide on a small number of _symbols_: I/Q patterns that will be used to represent data. Each symbol should be at least several samples long, and the symbols need to be easy to tell apart even when noise is added to the signal.
- Define an encoding: how bytes of data are mapped to sequences of symbols.
- Add a synchronization mechanism, e.g. a recognisable preamble to the transmission, so that the receiver can identify where each symbol starts and ends, and correct for any difference in clock rate between the transmitter and receiver.
- Add a checksum, or use an error correcting code, so that when symbols are misidentified due to noise, they do not result in misread data.
- Add filtering to keep the transmitted signal within an appropriate bandwidth.

Sasszem, 2023-06-23:
I suggest implementing an existing communication protocol first, for example RTTY. I also suggest not using phase-modulation, as decoding that often requires closed-loop frequency control (i.e. Costas loop), and FM-based modulations are a bit _simpler_ to decode.

straithe, 2023-07-03:
I'm closing this as there hasn't been a response in over 30 days from the original poster.

closed 2023-07-03 by straithe