greatscottgadgets/hackrf #1326: HackRF not detected via USB

oliverwolfat, 2023-05-18:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

HackRF One being detected by the host computer

### What outcome actually happened?

HackRF One not showing up in either Normal nor DFU mode.
Windows:
Not showing up in Device Manager neither in Normal nor in DFU mode

Kali Linux:
Not showing up in lsusb or dmesg neither in Normal nor in DFU mode

HackRF Board:
[image]

Normal Mode:
3V3, 1V8, RF are fully lit

DFU Mode:
[image]
3V3, RF fully lit
TX dimly lit

Revision r4_p1
MCU 
gets extremely hot

### What operating systems are you seeing the problem on?

Microsoft Windows 10 Pro 22H2
Kali Linux 6.1.0

### What is the output of ```hackrf_info```?

There is no Output, since the device is not detected

### Are you using any third-party software?

Not able to use any Software with it

### Are you using any third-party hardware?

No additions made

oliverwolfat, 2023-05-26:
I did some digging and found the troubleshooting steps from #684 
I measured the resistance on the micro USB connector with the following results.

Pin 1 (VCC) to Pin 5 (GND): 9.98 kOhm
Pin 2 (D-) to Pin 5 (GND): 1,623 MOhm
Pin 3 (D+) to Pin 5 (GND): 1,549 MOhm
Pin 4 (ID) to Pin 5 (GND): 10 kOhm

straithe, 2023-05-26:
How many different cables have you tried? https://hackrf.readthedocs.io/en/latest/usb_cables.html

oliverwolfat, 2023-05-27:
As stated in my original post, i tried 5 cables from 5 brands. 

I have a hackrf clone laying around as well, which has been detected by the host computer with all 5 cables.

I also tried three host pcs (2x Windows, 1x Linux) and different ports

oliverwolfat, 2023-05-27:
I have tried a sixth cable today, different style Micro USB to USB-C same results

straithe, 2023-05-27:
Are you connecting direct to the computer or through a USB hub?

oliverwolfat, 2023-05-27:
I am connecting directly via computer, but i also tried it via Hub. I can't say if it is powered or not, it has a power Delivery port (Anker USB-C Dock)

straithe, 2023-07-03:
I am curious and have one more question about the cables you've tried. Are they all micro to USB-C, or have you tried some that are micro to USB-A?

oliverwolfat, 2023-07-03:
Hi straithe,

I have tried about 7 different USB-A to Micro-USB cables (with and without the USB-C dock in between) and 2 different USB-C to Micro-USB cables (only without the USB-C dock) on 3 different devices.

KR
Oliver

straithe, 2023-09-13:
I suggest contacting your reseller to see if they will exchange your HackRF.

closed 2023-10-11 by straithe