greatscottgadgets/hackrf #1327: TCXO not used?

NemesisXB, 2023-05-19:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

I am new to hackrf. I bought a 2x hackrf one R9 and a TCXO for each of them. I installed the tcxo, but no matter what I do, the output of hackrf_debug --si5351c -n 0 -r is always 0x51. Also tried connecting the clock out of one the the clock in of the other. Still getting 0x51. Am I missing something?

### What are the steps to reproduce this?

1. Installed TCXO
2. Ran hackrf_debug --si5351c -n 0 -r 
3. Expect 0x01
4. Gets 0x51

jimmo, 2023-05-19:
@NemesisXB Maybe https://github.com/greatscottgadgets/hackrf/issues/1301#issuecomment-1492878894

NemesisXB, 2023-05-19:
> @NemesisXB Maybe [#1301 (comment)](https://github.com/greatscottgadgets/hackrf/issues/1301#issuecomment-1492878894)

Thought so, but I have one that seems to be confirmed working: https://github.com/greatscottgadgets/hackrf/issues/1301#issuecomment-1520547348 - I have this exact one (CLK-B)

jimmo, 2023-05-19:
> Also, I tried the clkout of one hackrf into the clkin of another. Still got a 0x51

Assuming you did, but just to be sure because you didn't say in the original message, did you enable the clock output on the source: `hackrf_clock -o 1`

Do you have a scope to verify the signal from the txco module?

NemesisXB, 2023-05-19:
I was actually busy looking for it. Where can I find windows builds for the tools? The pohtos sdr ones are from 2021 and that version of the hackrf_clock does not support the -i argument?

straithe, 2023-05-23:
There aren't pre-made Windows builds available at this time.

closed 2023-06-30 by straithe