greatscottgadgets/hackrf #1329: Opera Cake: Commands to switch port using Python

helmet21, 2023-05-20:
### What would you like to know?

Is there a way to control the HackRF with Opera Cake in Python? Actually there is the command line tool for the Opera Cake, but does anybody know how to change the antennas with Python not using the command line tool? What I actually want to achieve is to be able to change the antenna manually while I am getting samples.

 I know there is the time mode option but I'm having trouble synchronizing with the sampling process so it is unclear which samples belong to which port in the resulting data array. I want to be able to get 1000 samples, change the antenna with a command and then get another 1000 samples from a different port and repeat this with all four antennas.

martinling, 2023-05-20:
There are various implementations of a `pyhackrf` module out there to make the libhackrf API accessible through Python, but I don't think any of them have been updated to include the additional functions for controlling Opera Cake yet. You could look at extending one of them to add this - it might not take much work.

Time mode should be the right way to get what you want, though. It's the only way to ensure that the switchover happens after an exact number of samples, because it puts the timing under firmware control. If you use manual mode and request a switchover from the host whilst RX is ongoing, there's no way to control exactly when that will happen, or even to tell when it did.

What problems are you having with trying to use time mode?

helmet21, 2023-05-29:
Thanks for your reply, much appreciated.

The problem is knowing if the first sample i collect in my buffer is actually the first sample that the HackRF is transmitting. I.e. how do I know which samples belong to which port?

Thanks!

martinling, 2023-05-29:
The switching sequence starts from the beginning whenever the HackRF goes into TX or RX mode. So if you use, for instance, `hackrf_operacake -w 10 -t A1 -t A2 -t A3 -t A4` then each time you start RX, the first 10 samples you read will be from port A1, the next 10 from port A2, and so on.

This relationship should stay consistent as long as you keep reading out the data quick enough that the buffer on the HackRF doesn't fill up. To make sure that's not happening, you can use `hackrf_debug -S` after running your program to check if there were any overruns.

closed 2023-06-05 by martinling