greatscottgadgets/hackrf #1331: USB test points?

Commander-Crash, 2023-05-26:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I want to know where USB test points are located on the hackrf one board.

### What outcome actually happened?

To replace faulty microusb with USB C brakeout port

### What operating systems are you seeing the problem on?

Not Available this is hardware question.

### What is the output of ```hackrf_info```?

Not Available this is hardware question.

### Are you using any third-party software?

Not Available this is hardware question.

### Are you using any third-party hardware?

Not Available this is hardware question.

straithe, 2023-05-26:
There are no USB test points on the HackRF One board. The Micro USB connector is used.

closed 2023-06-09 by straithe

Commander-Crash, 2024-07-21:
To anyone looking for a way for USB out.
[image]