greatscottgadgets/hackrf #1333: Receiving and transmitting problem.

ArekPL77, 2023-05-28:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

N/A

### What outcome actually happened?

I have a problem that I cannot solve. Please advise.
I bought my first HackRf device a few years ago, it worked flawlessly until a path on the PCB from the antenna socket broke. I added a piece of wire between the antenna socket and the resistor on the PCB, but the device no longer worked as before. It transmits and receives with very low power. I have now bought a new device - and here's a surprise it transmits and receives just as poorly as the old damaged device.

I understand that a component in the old one was damaged, but why does the new one have the same problem? Please advise why this is so? 

What component was damaged in the old device? Can I replace it myself?

I am attaching pictures showing the remote control signal reception from the RTL-SDR and the damaged and new HackRF to show how poorly they receive signals. At HackRF I marked the signals with a circle, you can hardly see them.

[image]
[image]
[image]

### What operating systems are you seeing the problem on?

Windows 10

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

SDRSharp

### Are you using any third-party hardware?

original antenna

straithe, 2023-05-28:
Please add a picture of your board to this issue.

chris451gh, 2023-07-10:
Perform the following test: 
Tune to a strong signal. Enable 'amp'. If the signal does not increase, the mga amp is blown, and with that it takes out two of them also the output buffer amp, which cuts down the output power. This is caused by ESD or a voltage more than 4 volts on the antenna. There is a tvs diode on the input but it does not limit the voltage enough. Replace the buffer amps MGA81563. 
To prevent these from blowing out only operate with a limiter and dc block or a band pass filter, high pass filter or dc block.  I have 2 portapack mayhem's here and have changed these amps out twice on one unit and once on the other. Then, your amp(1) amp(0) setting will work and the signal will increase 12db when the amp is enabled. The reason TX is weak with a blown amp is it puts dc current on the RF switch which prevents it from working correctly. TX out will be down 20-30db and RX will be down 10-12db(with amp). 
[image]
[image]
[image]

ArekPL77, 2023-09-25:
Hi. I am sorry that I write back so late, but I had a contract of several months and I was not at home.
Thank you for your interest in the problem and reply to my post.
Could you get to me how to turn on 'AMP'?

straithe, 2023-09-25:
How you turn on the amo depends on the software you are using with your HackRF. Please consult that manual for the software.

chris451gh, 2023-09-25:
OK for most versions of hackrf mayhem, turn it on with a click of selector button,
for the H2. H1 I think has a separate button for power. If you see LED but no display play with R L U D Center while turning it on that enab les the display.
From on hit center and click receive, then select audio, pick Audio RX.
It says NFM 161.975 32 32 ____signal bar___ 84
Or whatever freq it was last on, or default freq.
Highlight the first 32.  Below appears amp 0, use the knob to set this to 1.
If the amp is working the waterfall will brighten up, and signal to noise on a known signal that its tuned to will go up 14 db. If the amp is no good it will drop.
if something else is also bad there will be no signals. In the US it is convenient to tune in national weather service 162.4-162.55, and you can tune to 500.000 you will see a strong signal with lots of sidebands, rotate amp on it should turn red over much of the center.
For transmit, tune to a frequency

chris451gh, 2023-09-25:
for transmit: turn on the Hackrf. Click to pass the logo page.
Here make note of the software version at the bottom and write down.
Select utilities. Select signal generator.
Set up a receiver on a test channel such as 435mhz.
set the transmit frequency there, I used 499.9mhz. Using touch screen set start
Adjust gain until the signal is weak to moderate on the receiver 0-45db
Now highlight amp and rotate knob, 0 or 1 for off and on.
It should be about 14db stronger with amp set to 1. If the output amp is good then
from 3 feet away the signal will be full scale and should be even stronger with amp on.
I got full scale on an Icom R20 in NFM mode 3 feet way with gain 0 amp 0.
It gets stronger by 14db(more quieting) with amp on.
If the amp is bad the signal may be -70dbm, very weak and not too good with gain up.
If the signal gets stronger as gain goes up the variable gain amp and the switch are good, but the swich chip does not work right with a blown amp, becuase the blown chip obstructs the signal from getting through the switch, I think because its putting load on the input output of the switch chip and it doesnt work the way it should when that happens.
The fix is to replace the surface mount amp chips. I never had to change the switch chip nor the variable gain amps. Just the MGA815 amps. I got a strip of 10 of them from Ebay. There is an equivalent chip by hittite that would work as a replacement. I have not tested that. I run a limiter with most antennas so 2 units continue to work fine with that sma limiter (mini-circuits). That keeps voltage and too much power off of the input and output. I think its ESD or ground loops that are enough to blow the preamp chips. I have had to replace the L band chips in Icom receivers because they got blown too.

straithe, 2023-09-25:
@ArekPL77 please note that @chris451gh's advice only works if you are using a Portapack and the Mayhem firmware. Are you using a Portapack?

ArekPL77, 2023-09-28:
I did as you advised.
I attach links to the movies I recorded.
    http://hostuje.net/file.php?id=351938364179ff8088f416adfe0916e7
    http://hostuje.net/file.php?id=0d8a052e8a0b7de6e3fcfdebb2ff4914
Could you judge if I have done your instructions well?

When enabling AMP in both cases, the signal clearly weakened. So it seems to me that MGA81563 elements are damaged.
I ordered them from Aliexpress, they should come in about 3 weeks. Should only these elements replace?

straithe, 2023-12-11:
It does seem as though your device has a hardware issue. Please contact the person you bought your HackRF + Portapack from for further support.

closed 2024-02-06 by straithe