greatscottgadgets/hackrf #1335: Hackrf sweep

lolakhairy, 2023-06-06:
### What would you like to know?

HI,
I need to run python script to plot the row data from sweep.csv file. the CSV file have those, Column. 
any help not sure how what to do with the last 5 column for db 

[image]

straithe, 2023-07-03:
Each of the columns represent the power level for a 1 MHz bin between 1 and 6 MHz. The first dB column is 1 to 2 MHz, the next is 2 to 3 MHz, then 3 to 4 MHz, and so on. You can use this information to:
- make a plot representing frequency in the x-axis if you plot a single row, or
- plot power against time for each column / bin

Drozdnik, 2024-07-08:
I have almost the same issue, as for me I got some unexpected behaviour , I run sweep with these settings:
hackrf_sweep -f 800:900 -w 50000 -r data800840.csv
But inside 5 MHz I get 808 dB values instead of 101. What I did wrong?