greatscottgadgets/hackrf #1337: hackrf_clock -i gives error message

zanco, 2023-06-16:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

I try to feed my hackrf with external 10 MHz source. I run the command hackrf_clock -i to check, but I get this error message:
hackrf_clock: symbol lookup error: hackrf_clock: undefined symbol: hackrf_get_clkin_status

hackrf_info (newer version HackRF board) shows:
pi@raspberrypi:~ $ hackrf_info
hackrf_info version: git-13dacd81
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc281b99c3
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x005d4758
hackrf_info: symbol lookup error: hackrf_info: undefined symbol: hackrf_board_rev_read

hackrf built from source on raspberry Pi

I had wired the same hackrf to an Ubuntu 22.04 system, enabled clock input and checked, which was OK, but on the Raspberry Pi I can't figure this out.  

Thing is, I run satellite observations in the 2200 MHz range, and they are about 8 kHz offset now. 

Thanks for your support,
Ben

### What are the steps to reproduce this?

Run the command as described in the issue explanation.

### Can you provide any logs? (output, errors, etc.)

hackrf_clock: symbol lookup error: hackrf_clock: undefined symbol: hackrf_get_clkin_status

martinling, 2023-06-16:
You're running a new `hackrf_clock` binary with an old version of `libhackrf`.

If you ran `make install` from the `host` directory in the source, you should already have a new version at `/usr/local/lib/libhackrf.so.0.8.0`. You probably just need to run `sudo ldconfig` to update the linker cache so that it finds the new version.

zanco, 2023-06-16:
Thanks for your quick reply. I ended with sudo ldconfig, but I entered host/build again, make clean, cmake .. followed by make, sudo make install and sudo ldconfig but the output from hackrf_clock -i stays:

hackrf_clock: symbol lookup error: hackrf_clock: undefined symbol: hackrf_get_clkin_status
Is there a way to check these with hackrf_clock -r ? I tried to find doumentation which clock number is which function but somehow I could not find that info.

martinling, 2023-06-16:
You can't use `hackrf_clock` at all without the up to date library.  A few things to try:

- Run `ldd /usr/local/bin/hackrf_clock` to see the libraries it's currently using.
- Check that the new library is installed in `/usr/local/lib`.
- Check whether `/usr/local/lib` is included in `/etc/ld.so.conf`.

zanco, 2023-06-16:
Thanks again. The result was that there was libhackrf.so.0.5.0 AND libhackrf.so.0.8.0 on the same system , the oldes being part of the Soapy driver installation. With some help I got instructions to change the logical link for libhackrf.so.0 to the libhackrf.so.0.8.0 and now the output for  /usr/local/bin/hackrf_clock -i  is:

CLKIN status: clock signal detected

So, two problems solved ;-)

zanco, 2023-06-16:
Solved by linking to the new library instead of the old 0.5.0 library

closed 2023-06-16 by zanco