greatscottgadgets/hackrf #1340: Phase sync coherence between two Hackrf One

ViniFS, 2023-06-27:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I was hoping for a time sampling GUI showing both HackRF One as RX sampling the same CW signal with synchronized samples of both IQs. A phase difference is OK but with constant jitter. It could be done by SMA clock or headers hardware triggering synchronization.

### What outcome actually happened?

When seeing the data of both HackRF One, one was being used as trigger so it's IQ was fixed on the same position in the visualization window, but the other one kept moving uncoordinatedly through the time domain. I have tried using on HackRF One to trigger the other one on both SMA connectors and also on headers pointed as hardware triggering, and I have tried also using an external square wave genrrator to trigger both of them. All these ways showed the same results.

### What operating systems are you seeing the problem on?

Linux 22.04

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

GNU Radio

### Are you using any third-party hardware?

N/A

straithe, 2023-10-19:
Thank you for opening this issue. Unfortunately, I'm not sure I understand what you need assistance with. I feel you may be trying to get help with GNU Radio, is that correct?

888XFX888, 2023-11-21:
> If you want fully synchronised sampling then you need to use _both_ mechanisms together: clock sharing using the CLKIN/CLKOUT SMA ports, and hardware triggering using the P28 header. They serve different functions. The shared clock is needed to keep the two sample rates the same, while the hardware triggering is needed to make both devices start sampling at exactly the same time.
> 
> For clock sharing, you can either connect the same 10MHz reference to the CLKIN port of both devices, or connect CLKOUT on one to CLKIN of the other. In the latter case, you need to enable the clock output on the first device with the `hackrf_clock -o` command.
> 
> Starting two captures at the same time can be arranged from the command line by using the `-H` option to `hackrf_transfer`, which will make the device wait for a trigger on the P28 header to start receiving. This can be used either to start one device at the same time as the other (with trigger output wired to trigger input), or to start both devices on an external signal (wired to the trigger input of both).
> 
> Unfortunately, I'm not sure it's possible to do hardware triggering through GNU Radio at the moment, as neither the OsmoSDR nor Soapy blocks appear to expose the necessary control.

hello  I also encountered this issue, but I used the method you mentioned to synchronize both hardware sampling and clock. However, after collecting the signal through gnuradio, I calculated the phase difference between the two hackrfs. This parameter is not stable and fluctuates in the range of - pi to pi. I am not sure what is causing this.

888XFX888, 2023-11-21:
I used a **2.4GHz RF signal source  connect to the RF input ports of two hackrfs after passing through the power divider.** 

At the same time, I also input the **same 10Mhz clock signal** on the clock input port. as well as  I **used the code**  `hackrf_ Transfer - r 1- d 600463dc2b426ec7- H 1 | hackrf_ Transfer - r 2- d 600463dc30a245c7- H 1 `Enable synchronous 。

Here is my GRC flowchart

[image]

But the phase I collected in this way is not stable,It will have a slow change。

[image]
[image]

 and **I want to know what is causing it**. Is there any other reason that can cause **unstable phase difference**

888XFX888, 2023-11-21:
This is my graduate project, and I really need your help. Thank you very much

martinling, 2023-11-21:
Hi @888XFX888,

The OsmoSDR source block is misleading you a bit here. It can be used with a lot of different devices but does not support all features on all of them.  Although it has options for clock source and time source, those settings do not do anything for HackRF. These options are currently only implemented in OsmoSDR for the USRP and BladeRF.

The external clock input is used automatically on HackRF if it is detected, so you do not need any setting in the GNU Radio source block to select this. However you should check that your clock input is suitable and that it is being detected. You can check this on the command line with `hackrf_clock -i`, which should display either `CLKIN status: clock signal detected` or `CLKIN status: no clock signal detected`. The clock signal should normally be a 3.3V square wave, though a sine wave is also OK. If the peak-to-peak voltage of the clock signal is less than 3V, whether it will work depends on which revision of the hardware you have.

Getting the two HackRFs to start sampling at exactly the same time (within 1 sample period) is not currently possible using GNU Radio. The simplest solution is to use `hackrf_transfer` with the `-H` option to capture the samples instead.  You can use the `-r <filename>` option on each `hackrf_transfer` command to tell it where to save the samples, and then use two `File Source` blocks in your GNU Radio flowgraph to read those samples, instead of using the OsmoSDR source. If you need to process in realtime, you could pipe each `hackrf_transfer` process to a TCP socket using `netcat` or a similar program, and use two `TCP Source` blocks in the flowgraph to receive the samples. Or, you could modify the [OsmoSDR code](https://github.com/osmocom/gr-osmosdr/blob/master/lib/hackrf/hackrf_source_c.cc) to make it support hardware sync, using the `hackrf_set_hw_sync_mode` function.

schneider42, 2023-11-21:
Is the hardware sync actually allowing for phase coherence between two HackRFs? I think that is what the OP wants to achieve. Start of sampling withing a certain time period is not good enough for them. They want a stable phase between the two devices between two runs.

martinling, 2023-11-22:
Hi @888XFX888,

That's a good trick you've found which I hadn't thought of. Because OsmoSDR never changes the hardware sync mode at all, it does indeed work to use `hackrf_transfer` with the `-H` option to enable hardware sync before starting GNU Radio.

Your time sink plot comes out wrong because the output format of `hackrf_transfer` is not floating point but rather signed 8-bit values directly from the ADC. You can use an `IChar to Complex` block with a scale factor of 127 to convert the data:

[image]

Could you please post the output of `hackrf_info` with both your HackRFs connected?

888XFX888, 2023-11-23:
Hi，@martinling.

Here is my hackrf_info :
[image]

martinling, 2023-11-23:
Hi @888XFX888,

Thanks for the info.

I wanted to see what hardware you were using, because the phase drift you see suggests that the clock synchronization is not working properly, and the possible causes for that depend on the hardware revision.

However, your boards were not manufactured by Great Scott Gadgets. So I cannot be sure of all the hardware details, and the boards have not been through our factory testing. There may be some hardware problem preventing the clock synchronization from working, but you would need to ask for support for that from the manufacturer or seller you bought them from.

It may be helpful for me to explain some background here about the external clock support.

In HackRF One revisions r1 to r8, the clock generator for the board is an Si5351C, which has built-in support for an external clock input. During the global chip shortage in 2020 onwards, the Si5351C part was not available, and we had to redesign the board to use the Si5351A instead, which does not have this feature. We needed to add some additional circuitry to enable support for an external clock input when using the Si5351A. This was done in the r9 version of the hardware design.

During the shortage, at least one manufacturer produced clones of the HackRF One that simply swapped the Si5351C for the Si5351A, without adding the additional circuitry. These clones did not work with an external clock at all, but were sold with the CLKIN port anyway. So in general I do not trust other manufacturers to ensure that this feature works properly.

Your boards identify as the r9 version, which means that in theory they should have the extra circuitry. And you say that the clock input is detected with `hackrf_clock`. But because the boards were not made by GSG, I cannot be sure that all the hardware is correct. The design may have been modified, or necessary components could have been substituted or left out.

888XFX888, 2023-12-07:
Hi，@martinling

It's been a while, and I finally found the problem with the unstable phase difference.

After replacing the received data cable, the phase difference between the two hackrfs received data became very stable.

I think it may be a problem with the data cable that causes unstable or unsynchronized data transmission, and these unstable or unsynchronized data can lead to problems with the final calculated phase difference.

So，There's no problem with Si5351A.

Thank you very much for your previous answer。

martinling, 2023-12-07:
Ah! That would make sense. If the USB link were unreliable due to a poor cable, then that would cause retransmissions to be needed, but at some point the buffer on the device will overflow and samples would have to be dropped, which would cause the two signals to get of sync.

This could have been detected using `hackrf_debug -S`  to check the shortfall counts, but I didn't think of it.

It seems that it would be useful to have a checklist for synchronization issues in the documentation. I'll open an issue for that.

martinling, 2023-12-07:
I've now put together a checklist in issue #1386 for troubleshooting synchronisation between multiple HackRFs, including all the points discussed here. At some point this should be transferred into the [HackRF documentation](https://hackrf.readthedocs.io/en/latest/).

I'm going to close this issue now since @888XFX888 has resolved their problem and there hasn't been an update from @ViniFS since their original post. If anyone's still having trouble after going through the checklist, please open a new issue and we'll do our best to help.

closed 2023-12-07 by martinling

AlMasz2837, 2024-03-07:
hi, is it possible to set the frequency of a rectangular pulse or submit it to hackrf.  frequency less than 10 MHz, for example 1 kHz, with an amplitude of 3.3 V?

Saiqin, 2024-07-19:
[image]

Hi all i do the same synchronization like above illustration, why i still have the delay for the two device.
thanks.

martinling, 2024-07-19:
@Saiqin please go through the checklist in #1386, and then open your own issue if you're still having problems.