greatscottgadgets/hackrf #1341: my old version hackrf not found tcxo

seniorjack, 2023-06-28:
### What would you like to know?

### **What issue are you facing?**

I have an hackrf board .when I got it ,it had a portapack， and then I uninstall the portapack screen and write a new firmware version 2023。

On this basis，I bought a tcxo and install it on my hackrf。 but it  seems doesn't work。

so i put the tcxo on another hackrf, it work。 what's the problem？it maybe the hardware or software problems?

**Here the hackrf_info information:**
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000629461dc23782d53
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00544f6c
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One

Ran hackrf_debug --si5351c -n 0 -r
Expect 0x01
Gets : [  0] -> 0x50

Ran hackrf_clock -i
CLKIN status: no clock signal detected

straithe, 2023-06-30:
Can you please add the output of ```hackrf_info``` from both HackRFs to this issue?

straithe, 2023-09-13:
Unfortunately there are some TCXO modules out there which are not compatible with the R9 hardware since they do not output the necessary voltage. We've always specified that a 3.3.V signal is required. Some modules were apparently being sold that didn't provide enough voltage, yet worked by chance with earlier HackRF One revisions.

straithe, 2023-10-11:
I'm closing this as there hasn't been a response from the original poster in over 30 days.

closed 2023-10-11 by straithe