greatscottgadgets/hackrf #1342: Anomalies in LNA and VGA sliders, ghost signals

YapperBot, 2023-07-02:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Adjusting the VGA past 30 in some programs results in a noise fall-off then resumes climbing up to max. LNA volume falls off at 24 then resumes climbing up to max. "Ghost" signals gradually shift from frequency to frequency as the frequency is adjusted. This is not DC. Known issues... the amplifier is blown. 

### What outcome actually happened?

N/A

### What operating systems are you seeing the problem on?

Hardware... firmware?

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

GNU Radio, SDRAngel, SDRTrunk

### Are you using any third-party hardware?

Portapack

straithe, 2023-07-03:
Can you please upload a picture of your HackRF?

YapperBot, 2023-07-03:
I can't, actually. I gave the device a look over and didn't see anything out of the usual. It's relatively new. I also reverted to the original firmware and the problem has persisted.

straithe, 2023-07-03:
Can you please send me a link to where you bought your HackRF?

YapperBot, 2023-07-03:
On AliExpress somewhere. The relevancy of this evades me.

straithe, 2023-07-03:
You mentioned in your initial issue statement that you believe you are experiencing a hardware or firmware issue. For Portapack firmware issues, you will need to contact the folks who make the Portapack firmware you are using. For hardware issues that are not on a Great Scott Gadgets HackRF, you will need to reach out to the person you purchased your hardware from. For software issues that are related to GNU Radio, SDRAngel, or SDRTrunk you will need to reach out to those communities.

YapperBot, 2023-07-03:
I've essentially just deduced that it's a hardware issue, and considering that all these devices use the exact same components regardless of manufacturer I'm beginning to feel pretty blue-balled that you're turning a cold shoulder to the matter despite likely having a pretty strong idea of what the problem is. Anyway, this'll be up for anyone else searching a similar HackRF issue to persist their frustrations over. G'day.

straithe, 2023-07-03:
Unfortunately we have found that the parts do differ across clones and it has hindered our ability to support those who haven't bought GSG devices.

closed 2023-07-03 by straithe