greatscottgadgets/hackrf #1347: HackRF one - GNU Radio on Windows - osmosdr_source block

lindnerlars, 2023-07-07:
### What would you like to know?

Hi community, my name is Lars.

I bought a HackRF One and playing around with GNU Radio on my Win10 machine.

The actual version 3.10.3.0 is missing the osmosdr_source block. What can you recommend? 🙂

Thank you very much and with best wishes, 
Lars

straithe, 2023-07-07:
Hi, Lars. I see that you've asked this question in the GSG Discord and have gotten responses there that would be the exact same information I would suggest here. (Download osmosdr blocks, and since you are on Windows try radioconda.) I'm closing this issue as it is a duplicate to the Discord conversation.

closed 2023-07-07 by straithe

lindnerlars, 2023-07-08:
Hi straithe, yes and sorry for my duplicated questions. What would be better, opening an issue here on github or asking in the discord channels? 

Thank you and have a nice weekend