greatscottgadgets/hackrf #1348: hackrf one firmware issue

afiqarifen, 2023-07-09:
### What would you like to know?

hi everyone.
i am using ubuntu 18.04 to update my hackrf one. I have updated my hackrf one using hackrf_spiflash -w hackrf_one_usb.bin. after i restart my device i check using hackrf_info. here are the output:

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc282a29c3
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x0061435b
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff

im sure my machine detect my hackrf one. but i cannot receive any signal from qspectrumanalyzer. 
ps. im not really sure whether its firmware issue or not.

thanks everyone

straithe, 2023-07-10:
To clarify, are you updating using apt-get, or are you updating using the code in this repository?

afiqarifen, 2023-07-10:
Im using apt-get to update. Is there any issue when updating using apt get?

Patnetpatnet, 2023-07-13:
I have the same problem. Updated to 2023 firmware with spiflash. Qspectrumanalyzer no longer shows any spectrum activity in both Ubuntu and Windows versions. SDR software is OK.

afiqarifen, 2023-07-13:
I app get remove the hackrf and build from the release. Now it can use. I think the issue with hackrf version.

dmaltsiniotis, 2023-07-13:
> Im using apt-get to update. Is there any issue when updating using apt get?

I think the issue is that the package maintainers have not updated the HackRF One libraries yet in upstream Ubuntu/Debian. Removing the package manager versions and building the latest from source will get you the latest libraries that should resolve the issue.

Patnetpatnet, 2023-07-13:
Hi, I have upgraded the firmware using the instructions here. I have confirmed the firmware is updated by running the info command. 

Please can you clarify. Are you saying this is wrong and I need to do something else? If so please can you explain what is wrong and how I go back. Can I just go back to an earlier firmware by following these existing instructions? 

https://hackrf.readthedocs.io/en/latest/updating_firmware.html

afiqarifen, 2023-07-13:
> I think the issue is that the package maintainers have not updated the HackRF One libraries yet in upstream Ubuntu/Debian. Removing the package manager versions and building the latest from source [...]

Thanks for the info. Its kind of weird because i have installed/uninstalled it couple of time using apt get. Most of the time the hackrf_info version shows 'git ver xxxxxx'

dmaltsiniotis, 2023-07-13:
I have heard of case where there were multiple versions of the libhackrf installed a system and one was taking precedent over the other and causing odd behavior. I can't say for certain if that is part of the issue here or not.

afiqarifen, 2023-07-13:
> Hi, I have upgraded the firmware using the instructions here. I have confirmed the firmware is updated by running the info command.
>
> Please can you clarify. Are you saying this is wrong and I need [...]

The updating command to update the firmare are not wrong. My issue is the hackrf_info and libhackrf version are not tally with the hackrf one firmware. Thats why my qspectrumanalyzer does not work. I think this also related to hackrf_sweep. After i build from source everything work great. 

https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html

I uninstalled the apt-get version and built it from source following the above instruction.

straithe, 2023-09-13:
@afiqarifen Checking in on this issue. I'm not certain but it seems your issue has been resolved. Is that correct?

Patnetpatnet, 2023-09-14:
Hard to say. Can't get Qspectrumanalyzer or pavsa's to work in Windows.
Qspectrumanalyzer functions Linux only

afiqarifen, 2023-09-14:
Initially, i am using Windows to test it out and it wont work, so i decided to use it on ubuntu instead. And same thing it does not work and decided to post it here. Will find the time and try the solution again by using Windows and give my outcome here

dmaltsiniotis, 2023-09-15:
> Initially, i am using Windows to test it out and it wont work, so i decided to use it on ubuntu instead. And same thing it does not work and decided to post it here. Will find the time and try the solution again by using Windows and give my outcome here

If using Windows, please be sure to use the latest libhackrf and hackrf_info that are built from source. The versions offered from Pothos SDR (which are the install instructions) are too far out of date for the latest firmware. If you'd like to use pre-built binaries of the latest HackRF libraries and tools that work with the latest firmware, I have compiled a release for them here: https://github.com/dmaltsiniotis/hackrf/releases/tag/v2023.01.1 Make sure to uninstall any existing versions before using this latest version, and qspectrumanalyzer should work again.

straithe, 2023-10-19:
@afiqarifen did the instructions from @dmaltsiniotis solve your issue?

straithe, 2023-12-11:
I'm closing this issue as there has been no response from the original poster in 2 months. Any other issues should be opened in a new GitHub issue.

closed 2023-12-11 by straithe