greatscottgadgets/hackrf #1349: HackRFone FW check out of the box

ktart92, 2023-07-13:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Expecting to see the hackrf firmware version being printed out. I ran through the instructions for setting up the HackRFone (https://www.hackers-arise.com/post/software-defined-radio-sdr-for-hackers-setting-up-your-hackrf-one) got through the sudo apt install hackrf portion just fine, but when i try to check the firmware with hack_rfinfo or sudo hackrf_info, i get the message below. This is a hackrfone straight out of the box and I can see the USB light on. Not sure if I'm missing other steps because it seemed to me that I should be able to see some FW version straight out of the box. 

I was able to make it "transmit" a signal below (TX light is on while running this), but the extra amplitude printout is concerning/I don't see an output on a spectrum analyzer. I was following this tutorial https://www.youtube.com/watch?v=3NWn5cQM7q4.
 
[image]

### What outcome actually happened?

hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.

### What operating systems are you seeing the problem on?

Windows 11 enterprise, running Ubuntu 22.04.2 LTS

### What is the output of ```hackrf_info```?

hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.

### Are you using any third-party software?

Ubuntu 22.04.2 LTS

### Are you using any third-party hardware?

No

dmaltsiniotis, 2023-07-13:
Hi @ktart92,

This is most likely happening because the host tools and libraries you have installed are out of date with the firmware you are using. Unfortunately, it seems as though the PothoSDR folks have not kept up with recent versions of the HackRF One tools and firmware, and their download uses a very old version of hackrf_info and libhackrf.

Also, it appears you're mixing Windows and Ubuntu. The HackRF libraries not compatible across Windows and WSL if you're trying to Windows Subsystem for Linux. I would recommend sticking with the native Windows tools, or only Ubuntu tools, but not both.

In any case, you'll need to compile/build the most recent versions of the tools. For Windows, your next logical question might then be: Where I can download the most recent HackRF tools and libraries for Windows? This is currently a topic of discussion within the community but the short answer is that direct downloads of the recent Windows tools are not quite available, yet. The good news is that you have some options to move forward:

1) If you're familiar with building source code on Windows, it's pretty straightforward to build the tools from source from this repository.
2) If building code from source isn't really your thing, I have provided the 2023.01.1 release of the HackRF tools available on a fork of this repository located here: https://github.com/dmaltsiniotis/hackrf/releases/tag/v2023.01.1 (I hope to contribute a build process that makes these automatically in the future to this official repository.)

Important note: Nor I, or the code release linked above, are affiliated in any with Great Scott Gadgets. I simply forked the repository and built the tools myself and created a Windows release so folks could have a way to download the most recent release binaries. Always use discretion when downloading and executing binaries files from unknown sources.

Thank you,

-DM

ktart92, 2023-07-17:
Thanks, I got around the error below. I managed to just follow the steps on this page: https://learn.microsoft.com/en-us/windows/wsl/connect-usb

hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.

Since I'm just reading a file and using it to transmit, there's no way that I can natively increase the output power is there (without the use of an external RF amplifier)?

straithe, 2023-10-11:
Hi, @ktart92 are you still experiencing this issue?

ktart92, 2023-10-12:
I’m ok now, thank you.

closed 2023-10-13 by straithe