greatscottgadgets/hackrf #1353: GSG PCB having extreme difficulties flashing. Only 3v3 is lit up, board my be bricked?

boredondrugs, 2023-07-30:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Tried to use the hackrf for the first time using a third party portapack with ANY operating firmware to no avail. 

### What outcome actually happened?

I read (to the best of my knowledge and capabilities which may be limited but still fairly competent) all of the FAQs, looked extensively at the open issues and made a pretty considerable effort at reading through closed ones also, but nothing quite matches my problem specifically. 

Yesterday after purchasing a brand new GSG hackrf r9 board from Canada Robitix listed on your website, I was making an attempt to flash the mayhem firmware on to the board using a Portapack h2+ PCB v3.6. I understand you guys aren't responsible for 3rd party technical issues, but I believe there's something wrong with the board, or perhaps has been since I purchased it. Before I got any further, I have spent tons of money on different cords. I have three in rotation, all brand new, various types. Ones with the ferrite core insulating it, another with a high speed usb. Unfortunately it's all going through an adaptor which shouldnt limit it's power or data loads, but unfortunately for the time being my ubuntu computer has a fried mother board and shes not doing well. 

I've tried multiple times to flash the device, with and without the portapack attached. The last time I did it the lights all turned off as if it was in. DFU mode, the problem being is in terminal I'm not seeing that my computer is recognizing the device. At all, whether it's using homebrew, or hackrf tools, nothing seems to want to connect. when power is supplied to the device, the 3v3 light pops on and it does nothing. 

I would really appreciate any help or guidance through this. I've bought two of these so far and each one has failed and either bricked or died. 
Using the DFU boot: 
peterbachman@Peters-MBP ~ % dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Could not open file hackrf_one_usb.dfu for reading: No such file or directory
(The directories do exist and have been unzipped)

Other commands:
peterbachman@Peters-MBP ~ % hackrf_spiflash -s
hackrf_open() failed: HackRF not found (-5)
peterbachman@Peters-MBP ~ % hackrf_spiflash -R
hackrf_open() failed: HackRF not found (-5)
peterbachman@Peters-MBP ~ % 

### What operating systems are you seeing the problem on?

MacOS, Ventura 13.4.1

### What is the output of ```hackrf_info```?

peterbachman@Peters-MBP ~ % hackrf_info
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
No HackRF boards found.
peterbachman@Peters-MBP ~ % 

### Are you using any third-party software?

Tried to download gnu radio but it hasn't worked so far. It just keeps crashing. 

### Are you using any third-party hardware?

Yes, the Portapack H2+ PCB v3.6

martinling, 2023-07-30:
Please start by removing the PortaPack from the HackRF, as it can interfere with the HackRF starting up correctly.

Then boot the HackRF into DFU mode, by holding the DFU button while pressing the reset button.

When the HackRF is in DFU mode, if you run `lsusb`, you should see the DFU device listed as `1fc9:000c NXP Semiconductors LPC`. Make sure you see this first.

Once you can see that device, then you can use `dfu-util` to load the firmware into RAM. However, you need to give `dfu-util` the correct path to the file.

The message `dfu-util: Could not open file hackrf_one_usb.dfu for reading: No such file or directory` just means that `dfu-util` couldn't find the filename you gave it. This doesn't indicate any hardware issue. The program hasn't even tried to find the device at that point.

You need to either first change to the directory the file is in with `cd`, or give it the path to find the file. For instance:

```
$ cd ~/Downloads/hackrf-2023.01.1/firmware-bin
$ dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu
```

boredondrugs, 2023-08-02:
Hey Martin, I really appreciate the help, but it’s not even coming up when I use the lsusb command. Normally I’ve seen it say 1fc9:000c NXP Semiconductors when I open up system settings but regardless of what I do to access the board nothing is happening with it. I’ve used windows, Ubuntu, and Macos (the latter primarily) and still nothing is recognized. The 3v3 light will faintly come on and that’s the only function it currently has. Could the board be faulty? I haven’t even been able to load it on to gnu radio (mostly because macos seems to hate using non-native apps) and I just purchased it like two weeks ago. I went with the official gsg board because I didn’t think the Chinese pcb would be as effective or accurate but ironically it’s the only one I’ve been able to use. Any other suggestions?

boredondrugs, 2023-08-02:
Oh shit, it actually worked after trying it again. I'm sorry, I feel
like a total idiot. but here's the weird thing, is this normal? This is
after I've put the portapack firmware on the divice. Although you guys
may not do anything with that. 

I got this when I typed in hackrf_info:

***@***.*** ~ % hackrf_info
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: RunningFromRAM
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00744762
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
***@***.*** ~ % 

So it's running off ram. I'll check to see what I can do about this. But
apparently the upload was a success.

martinling, 2023-08-02:
That all looks good. Currently that firmware is running from RAM after being temporarily loaded by `dfu-util`. To update the firmware in flash you can then use `hackrf_spiflash`. If you flash the `hackrf_one_usb.bin` file from the current 2023.01.1 release, your HackRF should then work normally. If you want to use a Portapack firmware you can flash that instead, but please make sure that the firmware you use is compatible with both the HackRF r9 hardware and with the Portapack that you're using.

boredondrugs, 2023-08-02:
I got it!! Thank you. Sent from my iPhone

martinling, 2023-08-02:
Great!

closed 2023-08-02 by martinling