greatscottgadgets/hackrf #1355: Problem with osmocom-sdr and Hackrf with raspberry pi 4.

Rashbond, 2023-08-07:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Running simple script with osmocom source and frequency sink for detecting LoRa packets.

### What outcome actually happened?

<<< Welcome to GNU Radio Companion 3.9.8.0 >>>

Block paths:
	/home/pi/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/pi/Downloads/test1.grc"
>>> Done

Generating: '/home/pi/Downloads/test1.py'

Executing: /usr/bin/python3 -u /home/pi/Downloads/test1.py

/home/pi/Downloads/test1.py:12: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.version import StrictVersion
Traceback (most recent call last):
  File "/home/pi/Downloads/test1.py", line 247, in <module>
    main()
  File "/home/pi/Downloads/test1.py", line 225, in main
    tb = top_block_cls()
  File "/home/pi/Downloads/test1.py", line 46, in __init__
    gr.top_block.__init__(self, "Recivier_test1", catch_exceptions=True)
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gr/top_block.py", line 93, in __init__
    self._impl = top_block_pb(name, catch_exceptions)
ValueError: pmt_dcons: not a dict: wrong_type #<unknown>

>>> Done (return code -11)

### What operating systems are you seeing the problem on?

Ubuntu 22.04 LTS (On Raspberry pi 4)

### What is the output of ```hackrf_info```?

hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: XXXXXXXXXXXXXXXXXX
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: XXXXXXXXXXXXXXXXX

XXXXXXXXXXXX -  For Security Purpose

### Are you using any third-party software?

GNU Radio 3.9.0 with gr_LoRa cpp0

### Are you using any third-party hardware?

N/A

straithe, 2023-10-11:
It looks like your HackRF firmware is out of date. I suggest updating that. As for GRC and LoRa, I am not as familiar with them. You may find more help in the [GSG Discord](https://discord.gg/rsfMw3rsU8) or from the GNU Radio [website](https://www.gnuradio.org/) or community.

straithe, 2023-12-11:
I'm closing this issue as there has not been a response from the original poster in over 50 days.

closed 2023-12-11 by straithe