greatscottgadgets/hackrf #1357: Is it possible to change the transmission frequency during broadcasting?

ZxUkr, 2023-08-19:
### What would you like to know?

Can I use hackrf_set_freq() function inside receive/transmit callback function for change carrier frequency for all next data during broadcast?
Will it be work correct?

ZxUkr, 2023-09-24:
So only exist way to change transmission frequency is throuth:
0) flush_callback
1) hackrf_stop_tx
2) hackrf_set_freq
3)hackrf_start_tx
?

Is it exist more fast way to change frequency?

VioletGiraffe, 2023-10-17:
I am also interested in the fastest way to change the frequency during transmission, can we please get an official, developer-approved canonical answer / algorithm? Thanks!

VioletGiraffe, 2023-10-21:
Great answer, appreciate it. So there's no problem with changing the frequency while transmission is running, only with changing it from the async callback call stack. That is, apart from the reasons why one might not want to change it on the fly at all.

closed 2023-10-21 by martinling