greatscottgadgets/hackrf #1358: Richard Neese

RichNeese, 2023-08-23:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

dfu-util: Warning: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release

dfu-util -V
dfu-util 0.11

### What are the steps to reproduce this?

dfu-util --device 1fc9:000c --alt 0 --download firmware-bin/hackrf_one_usb.dfu
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1fc9:000c
Device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        42564 bytes
Download done.
dfu-util: unable to read DFU status after completion (LIBUSB_ERROR_IO)
root@pinebook-pro:/usr/src/hackrf-2023.01.1# hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf 
File size 37629 bytes.
LED1/2/3 blinking means CPLD program success.
LED3/RED steady means error.
Wait message 'Write finished' or in case of LED3/RED steady, Power OFF/Disconnect the HackRF.
Write finished.
Please Power OFF/Disconnect the HackRF.
root@pinebook-pro:/usr/src/hackrf-2023.01.1# hackrf_info
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000321864c8383f3a1d
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x006a474c

### Can you provide any logs? (output, errors, etc.)

no logs to post

RichNeese, 2023-08-23:
so it has updated the BL and other info but not the firmware itself

RichNeese, 2023-08-23:
screenfetch
         _,met$$$$$gg.           root@pinebook-pro
      ,g$$$$$$$$$$$$$$$P.        OS: Debian 12 bookworm
    ,g$$P""       """Y$$.".      Kernel: aarch64 Linux 6.4.10-edge-rockchip64
   ,$$P'              `$$$.      Uptime: 17h 0m
  ',$$P       ,ggs.     `$$b:    Packages: 3072
  `d$$'     ,$P"'   .    $$$     Shell: bash 5.2.15
   $$P      d$'     ,    $$P     Resolution: 1920x1080
   $$:      $$.   -    ,d$$'     DE: Xfce
   $$\;      Y$b._   _,d$P'      WM: Xfwm4
   Y$$.    `.`"Y$$$$P"'          GTK Theme: Numix [GTK2]
   `$$b      "-.__               Icon Theme: Adwaita
    `Y$$                         Font: Not Identified
     `Y$$.                       Disk: 19G / 60G (33%)
       `$$b.                     CPU: ARM Cortex-A53 Cortex-A72 @ 6x 1.416GHz
         `Y$$b.                  GPU: Mali-T860 (Panfrost)
            `"Y$b._              RAM: 1587MiB / 3863MiB

martinling, 2023-11-01:
Sorry, this seems to have been missed.

It looks like you were following some wrong or outdated instructions.

The `dfu-util` step is only needed to recover a board with bad or no firmware. It only loads new firmware temporarily into RAM. To write firmware to flash, you need to use the `hackrf_spiflash` tool. It's also no longer necessary to use the `hackrf_cpldjtag` tool. Current versions of the firmware include the CPLD image in the firmware binary, and load it themselves at startup. The contents of the CPLD flash on the board no longer matter.

The current instructions for firmware update are [here](https://hackrf.readthedocs.io/en/latest/updating_firmware.html). Since your board seems to be working with the 2015.07.2 firmware, the only command you should need to update it is `hackrf_spiflash -w hackrf_one_usb.bin`, run from the `firmware-bin` directory of the current release.

closed 2023-12-19 by straithe