greatscottgadgets/hackrf #1359: 2021.03.1 vs 2022.09.1 (or later)

voxo22, 2023-09-07:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Hello, I have a problem with software I use for DAB+ broadcasting (https://www.opendigitalradio.org/odr-mmbtools-releases). I use 2 hackrfs, one of them with the latest FW release on 2023.01.1, connected to Linux machine.

I have found that from FW version 2022.09.01 (and later 2023.01.01) the broadcast signal is accompanied by nonstop but random short dropouts (around 50 to 100 ms), constellation errors, to which the DAB receiver responds "out of SYNC". The received programme is therefore constantly interrupted and cannot be listened to at all. As soon as I returned back to 2021.03.01 (API 1.04) FW version, everything goes smoothly without any dropouts. Tested on two different HackRF boards. I am going crazy about this. Can you help, please?

hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000048866dc2f3e5dc3
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x0054476a
Hardware Revision: older than r6
Hardware supported by installed firmware: HackRF One

### What are the steps to reproduce this?

install odr-mmbtools, configure and start DAB broadcast. Listen on any DAB radio.

### Can you provide any logs? (output, errors, etc.)

no specific logs available, I have no idea where to search, everything seems to be running ok.

martinling, 2023-09-07:
I wonder if this is related to underruns in the data stream from the host. 

The 2022.09.1 release changed how the firmware behaves in the case of TX buffer underruns, following the the merge of PR #982, which aimed to fix #180 by ensuring that nothing further is transmitted if the USB stream stops.

Previously, if there was an underrun then the device would just repeat the last data in its TX buffer until new data arrived. The result would have been duplicate/invalid data in the transmission, but the accidentally repeated "filler" might have been enough to make a DAB receiver keep trying to decode. If underruns were short, then reception could recover without any noticeable glitch in the audio.

Things to try:

1. If you run `hackrf_debug -S` immediately after running your transmitting software, does it report any shortfalls?

2. If you run `hackrf_debug -T 1` before running your transmitting software, does transmission stop completely?

voxo22, 2023-09-07:
Hello,
ok, debug -S cannot be run after start, because of "hackrf_open() failed: Resource busy (-1000)". Just after stop trasmitting is reports: M0 state:
Requested mode: 3 (TX_START) [complete]
Active mode: 4 (TX_RUN)
M0 count: 398696448 bytes
M4 count: 398696448 bytes
Number of shortfalls: 132
Longest shortfall: 59401728 bytes
Shortfall limit: 0 bytes
Mode change threshold: 0 bytes
Next mode: 3 (TX_START)
Error: 0 (NONE)

debug -T 1 causes no transmission at all, probably due to immediate underrun

martinling, 2023-09-07:
That confirms that the issue is with the host not keeping up with supplying samples to the HackRF.
Unfortunately it was just by chance that the old firmware behaviour hid this problem.

Some things you could try:
- Make sure the HackRF is not sharing a USB bus with other devices.
- Reduce sample rate if possible.
- Increase the scheduling priority of the process running the transmission.
- Try a different host computer or OS.

closed 2023-09-07 by voxo22