greatscottgadgets/hackrf #1360: QAM complexity

voxo22, 2023-09-07:
### What would you like to know?

Hello, I have a quite simple question:
To what complexity is it possible to record a QAM signal via hackrf_transfer? Is 8-bit I/Q resolution enough for 256QAM, or even 512QAM?
I recently tested DVB-T2 256QAM rx/tx via transfer and it looks like the signal (8 MHz BW) is recorded/played without problems. I've also tried gnu-radio 256QAM DVB-T2 broadcasts and everything looks ok. However, why doesn't the rx/tx signal work for cable-TV (DVB-C) with the same 256QAM / 8 MHz BW? Is such a signal significantly different from DVB-T2 of the same complexity, or is the problem somewhere else? What about 512QAM and higher?
Thank you

martinling, 2023-09-07:
It will depend on signal quality, symbol rate, how accurately you can keep the gain set, and other factors.

8-bit IQ sampling means that you have a 256x256 space of possible sample values, and 256-QAM is a 16x16 constellation of points. So in principle yes, you can use one to capture the other.

But unless the RX gain is set perfectly and the sampling is perfectly locked to the phase of the signal, the constellation is going to be scaled and rotated.

Imagine you're trying to film a 16x16 display with a 256x256 pixel camera, while everything is spinning and zooming in and out. That's what you're up against.

voxo22, 2023-09-07:
ok, I understand. What I do not understand is the DVB-C standard, which is impossible to replay. Receiver always says; Signal power 100%, quality 0%, no matter how I set input parameters in Rx way, no matter frequency carrier. However, I have no doubts about source signal quality in the cable, should be brilliant.

martinling, 2023-11-01:
I'm closing this issue as we can't really provide detailed support on replaying specific signals.

What I will say though is that `hackrf_transfer` is not intended to be a replay tool. It's just a low level utility for transmitting and receiving raw samples from the HackRF.

To get a clean, well-scaled capture you really need to be using some visualisation and signal processing tools, typically through GNU Radio, so that you can see your target signal in the frequency and time domains, make sure it's not affected by the DC spike, set the RX gain to the optimal level, filter out adjacent signals, etc.

I recommend watching episode 11 of our [SDR video course](https://greatscottgadgets.com/sdr/), which covers replay specifically. It uses much simpler signals as an example, but all the issues it talks about apply for more complex modulations like DVB-C as well.

closed 2023-11-01 by martinling