greatscottgadgets/hackrf #1361: !!!Can some one tell me. My hackrf one rx and usb are glowing . the device is not getting detected. I even tried changing many usb cable but none of them could solve. 2 months early hackrf  was working perfectly fine. but suddenly the sdr is not detecting it . Only 3v3, 1v8 and rf led is glowing. Please help me to figure it out the issue

kartikhilare21, 2023-09-15:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

the device is not getting detected. I even tried changing many usb cable but none of them could solve. 2 months early hackrf  was working perfectly fine. but suddenly the sdr is not detecting it . Only 3v3, 1v8 and rf led is glowing. Please help me to figure it out the issue

### What outcome actually happened?

 Only 3v3, 1v8 and rf led is glowing.  usb and RX are not glowing

### What operating systems are you seeing the problem on?

win 11

### What is the output of ```hackrf_info```?

showing hackrf_info command not found

### Are you using any third-party software?

sdr sharp and sdr++ 

### Are you using any third-party hardware?

sdr sharp and sdr++

straithe, 2023-09-16:
Moved information from title to here: 

"!!!Can some one tell me. My hackrf one rx and usb are glowing . the device is not getting detected. I even tried changing many usb cable but none of them could solve. 2 months early hackrf  was working perfectly fine. but suddenly the sdr is not detecting it . Only 3v3, 1v8 and rf led is glowing. Please help me to figure it out the issue."

straithe, 2023-10-11:
Are you using a virtual machine or USB hub, @kartikhilare21 ?

straithe, 2023-11-28:
I'm closing this issue as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2023-11-28 by straithe