greatscottgadgets/hackrf #1369: Verifying external clockinput and 1pps on HackRF SDR

swapnachallagundla, 2023-10-31:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

How to test if the external 10MHz and 1pps inputs are activated or not on Hack Rf radio?

### What outcome actually happened?

I want to synchronize multiple hackrfs installed at different places. I am using GPSDO for generating 10MHz clock input and 1 PPS  for frequency and time synchronizations. I connected 10MHz at CLOCKIN input and 1pps at pin 16 of P28 on HackRF. Now I want to test whether these two signals are actually detected or not. 
I used the command "hackrf_clock -i" to check it. But I did not get anything.

I also tried to use this command  "hackrf_debug -s -n 0 -r" in terminal during transmission/reception. But it displaying the message hack rf already in use.

### What operating systems are you seeing the problem on?

Linux

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

N/A

mmm545, 2023-10-31:
> I also tried to use this command "hackrf_debug -s -n 0 -r" in terminal during transmission/reception. But it displaying the message hack rf already in use.

Did you try running the command while NOT receiving/transmitting? Because that's what you're supposed to do.

swapnachallagundla, 2023-10-31:
Yes, I tried. It returned 0x51 when I run the above command. 

Do I need to run any commands after connecting these external inputs?

swapnachallagundla, 2023-10-31:
When I run the command hackrf_debug --si5351c -n 0 -r on r6 model it displays 0x01

and when I run the same command on r8 and r9 models it displays 0x51.

Also, hackrf_clock -i is not working and it displays invalid option when I run this command.

martinling, 2023-10-31:
You should use `hackrf_clock -i` to check for the presence of the external clock input. This needs to be run when the HackRF is not in use by other software. If a clock is detected on CLKIN, it will be used automatically when TX/RX is started.

The old way of checking this using `hackrf_debug` is deprecated, and does not work with all hardware revisions.

You can check that the 1PPS input is working by running `hackrf_transfer` with the `-H` option for hardware triggering:

`hackrf_transfer -r /dev/null -H` 

This should display `Waiting for trigger...` and then start reception on the next 1PPS pulse.

martinling, 2023-10-31:
> Also, hackrf_clock -i is not working and it displays invalid option when I run this command.

Then you should update your host tools to a newer version, which will have this option.

swapnachallagundla, 2023-10-31:
We are using the latest version r9 of hackrf which is updated firmware.

martinling, 2023-10-31:
Yes, but if your `hackrf_clock` does not have the `-i` option then your host software is older than that. That option was added #1135 and included in the 2022.09.1 release.

swapnachallagundla, 2023-11-01:
Thanks.
 I have updated host software. Now, I am able to detect clock and 1pps is also working.

martinling, 2023-11-01:
Great! Glad we could help.

closed 2023-11-01 by martinling