greatscottgadgets/hackrf #1370: Is This an Authentic HackRF One Unit?

AlexLandherr, 2023-10-31:
### What would you like to know?

I’ve been looking for local sellers of a HackRF One unit/kit and found this one:
[https://fyndiq.se/produkt/hackrf-one-1mhz-till-6ghz-sdr-platform-software-defined-radio-339a0707e8cd427a/](https://fyndiq.se/produkt/hackrf-one-1mhz-till-6ghz-sdr-platform-software-defined-radio-339a0707e8cd427a/)

Based on the available images could this be a legitimate case? I haven’t seen one case such as this before which is why I’m asking.

I apologize if this question is out of place for this category but I would like to have some input before buying.

martinling, 2023-11-01:
That case is not one that we manufacture, and the seller is not one of our resellers, so I think that it's very unlikely that it will come with an original HackRF One board. The case design looks similar to some that are available from AliExpress.

We do have one reseller, NooElec, that has manufactured their own aluminium case which they offer as option with an original HackRF One board, but they are the only supplier doing this that I am aware of.

If you're looking for a source in Sweden then NooElec do also sell through Amazon.se, they have a bundle [here](https://www.amazon.se/SMA-adapterpaket-Inkluderar-frekvensomr%C3%A5de-bandbredd-SMA-adaptrar/dp/B01K1CCHR0/) with our ANT500 antenna, adapters and the original plastic case; or [here](https://www.amazon.se/NooElec-HackRF-komplett-paket-SMA-antennadapterpaket/dp/B07VZX2D2W/) with a TCXO add-on and their custom aluminium case as well.

The latter was the kit I bought for myself, a couple of years before later joining GSG, and I've been very happy with it.

AlexLandherr, 2023-11-01:
Thanks for answering! I asked the fyndiq.se seller and I presume they will answer the same or similarly.

AlexLandherr, 2023-11-01:
> That case is not one that we manufacture, and the seller is not one of our resellers, so I think that it's very unlikely that it will come with an original HackRF One board. The case design looks [...]

Is there any kit that has just the metal case and TCXO?

martinling, 2023-11-01:
They do custom kits through their own site [here](https://www.nooelec.com/store/sdr/sdr-transceivers/hackrf-162.html).

AlexLandherr, 2023-11-01:
One final follow-up question; are there any good images of the pinout for the connectors that allows for the TCXO connection?

When I read on Nooelecs's page it just says:
```
Plug & Play: just plug into header P22!
```

I need some diagram like the ones for the Raspberry Pi's GPIO pinout that has some labels so I can easily insert the TCXO without risking an electrical problem once the HackRF One is plugged in.

I've checked on the official documentation page under "Expansion Interface" but it's just a list of what the pins are for.

martinling, 2023-11-01:
On their TinyTCXO module it actually is that simple, because the position where it should fit the HackRF P22 header is marked on it, like this: 
[image]

AlexLandherr, 2023-11-01:
Would this video be accurate too in terms of where on the connector?:
https://m.youtube.com/watch?v=0WWZDptIoM0

martinling, 2023-11-01:
Yes, that's right.

AlexLandherr, 2023-11-01:
Thanks! Will order one soon from one of your listed sellers since the creators deserve to be funded.

closed 2023-11-01 by AlexLandherr