greatscottgadgets/hackrf #1374: Is This A Useable External OCXO?

AlexLandherr, 2023-11-05:
### What would you like to know?

I’m wondering whether this OCXO would be a viable one for my HackRF One?:
[https://www.amazon.se/Frequency-Standard-Reference-Oscillator-Ljudsystem/dp/B0B5HG23T3](https://www.amazon.se/Frequency-Standard-Reference-Oscillator-Ljudsystem/dp/B0B5HG23T3)

I have the Nooelec “Tiny TCXO” but would eventually like to use something better than it.

martinling, 2023-11-06:
It's not clear from that listing what the output waveform is. You want a 0 to 3.3V square wave.

See the [clock interface documentation](https://hackrf.readthedocs.io/en/latest/external_clock_interface.html).

AlexLandherr, 2023-11-06:
Ok, I'll stay clear of that particular one.

If any compatible product comes to mind please share it with me here.

martinling, 2023-11-06:
A good TCXO will be more than stable enough for the vast majority of purposes. Unless you have a very specific reason to need the higher stability of an OCXO, I wouldn't worry about looking for one. You will not notice any difference.

AlexLandherr, 2023-11-06:
Ok, thanks! Now I know.

closed 2023-11-07 by AlexLandherr