greatscottgadgets/hackrf #1375: Firmware update successful, firmware lost after powercycle

shnizzl, 2023-11-06:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

Hello,
a friend gave me a old HackRF One he described as "bricked", without investing too much time in troubleshooting. He said he attempted to update the firmware, which apparently failed.

Powering on the HackRF leads to only the 3,3V LED to light up. In this condition it is not recognized via _hackrf_info_ or _lsusb_. It looks like DFU mode, but I am unable to flash the firmware via dfu-util.

However, if I hold the DFU button and press the reset button to enter DFU mode "again", I am able to flash the firmware via dfu-util and hackrf_spiflash. After flashing, everything seems to be okay, except from the serial number listed via the _hackrf_info_ command being "RunningFromRAM". After resetting the HackRF the 3,3V LED lights up again, and the HackRF can not be found via _hackrf_info_. See the complete output of the commands below. This is reproducible. 

`dfu-util --device 1fc9:000c --alt 0 --download firmware-bin/hackrf_one_usb.dfu
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1fc9:000c
Device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        42564 bytes
Download done.
dfu-util: unable to read DFU status after completion (LIBUSB_ERROR_IO)`
`hackrf_spiflash -w firmware-bin/hackrf_one_usb.bin
File size 42500 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 42500 bytes at 0x000000.`
`hackrf_info
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: RunningFromRAM
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x003f4362
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One`

### What outcome actually happened?

I expected the firmware to be stored "persistent" in the HackRFs memory, being able to boot up in the working condition it has right after flashing the firmware. 

### What operating systems are you seeing the problem on?

Debian 12 / macOS 14

### What is the output of ```hackrf_info```?

`hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
No HackRF boards found.`

or

`hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: RunningFromRAM
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x003f4362
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One`

### Are you using any third-party software?

No

### Are you using any third-party hardware?

No

martinling, 2023-11-06:
It's possible some protection bits have been set on the flash chip.

Please post the output of `hackrf_spiflash -s -v` to check.

You can also try adding the `-c` option when you run the flashing step:

`hackrf_spiflash -c -w firmware-bin/hackrf_one_usb.bin`

shnizzl, 2023-11-06:
@martinling Thanks for your answer. This is the output of `hackrf_spiflash -s -v`:
```
hackrf_spiflash -s -v
SRP0	0
SEC	1
TB	1
BP	4
WEL	0
Busy	0
SUS	0
CMP	1
LB	0
Res	0
QE	1
SRP1	0
```

Flashing the firmware with the `-c` option did the trick. After reseting via button or power cycling the HackRF responds to `hackrf_info` with 
```
hackrf_info
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000469[.......]
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x003f4362
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
```

Running `hackrf_spiflash -s -v` again shows that the bit values changed:
```
hackrf_spiflash -s -v
SRP0	0
SEC	0
TB	0
BP	0
WEL	0
Busy	0
SUS	0
CMP	0
LB	0
Res	0
QE	1
SRP1	0
```

Thanks again, you really saved my day!

closed 2023-11-06 by shnizzl