greatscottgadgets/hackrf #1378: Correct Driver for SDR# on Windows 10?

AlexLandherr, 2023-11-11:
### What would you like to know?

I’m in the process of setting up my HackRF One for Windows 10 with SDR#, WXtoImg and Orbitron.

My SDR# version is `v1.0.0.1361`; which is the version that worked with the “VB-Cable” audio driver for piping audio from SDR# to WXtoImg.

I think I have the right driver for the HackRF One based on that SDR# is receiving data and the FFT and waterfall diagrams are updating.

What worries me is the indicator lights on the HackRF One seen below:
[image]

Does the indicator lights seem okay to someone more experienced?

And what Zadig shows me:
[image]

I’m only using it for reception of APT signals (137 MHz).

My antenna is the “V-dipole” kit from RTL-SDR.

martinling, 2023-11-11:
You have 3.3V (green), 1.8V (yellow) and RF (red) LEDs lit indicating that the three power supplies are on, you have the USB (green) LED indicating that it's connected to the host, and you have the RX (yellow) LED indicating that it's in receive mode. That all looks correct.

If you press the stop button in SDR# then you should see the RX LED go out. If you transmit then the TX LED (red) will come on. Aside from that you should not normally see the others change.

However, I think you may have missed a little assembly detail. If you find the original plastic case that came with your HackRF One, inside it will be a transparent plastic piece that goes over the LEDs. You can take that out and put it in the aluminium case - there is a recess in the end panel that it fits into. It's not essential, but it helps prevent the light from the LEDs leaking over to the adjacent holes, which makes it clearer which are lit and which are not.

The Zadig screenshot looks fine, but note that there should be no need to use Zadig at all. Any instructions suggesting you need to use it are long out of date. Since 2016 the firmware includes a WCID descriptor which tells Windows to use the WinUSB driver, so everything is automatic. This is indicated by the green tick and WINUSB text in the Zadig screenshot.

AlexLandherr, 2023-11-11:
Ok, thanks!

I tried for almost a week to find anything current for 2023 or so. Did not find much.

I do have a few follow-up questions if that's okay:

1. Is there any modern way for me to connect SDR# with Orbitron so that SDR# can receive Doppler corrections?
For some older versions of SDR# there's plenty of instructions that use a plugin called "SatelliteTracker2" for which I have the files. I have no idea if they work with modern SDR# releases.

2. Due to the layout of my house I have to use an SMA cable that is a few meters long (that is two connected ones from separate RTL-SDR V-dipole antenna kits).
This makes the signal received by HackRF look somewhat weak (poor SNR I think is the right term).
Other than shortening the SMA cable length what can I do to improve the SNR situation?

closed 2023-11-18 by AlexLandherr