greatscottgadgets/hackrf #1380: Are There Any Special Setup Procedures for Nooelec Tiny TCXO On Debian 12?

AlexLandherr, 2023-11-18:
### What would you like to know?

I’m preparing to test my HackRF One unit with my very capable Debian 12 laptop and am wondering besides running `sudo apt install hackrf’ is there anything else I have to do since I have installed/mounted the “Tiny TCXO” from Nooelec?

I plan on using gqrx as my SDR software.

straithe, 2023-11-28:
What happens when you run "hackrf_info" on command line when your HackRF is plugged in?

AlexLandherr, 2023-11-28:
I askes Nooelec and they said I could just plug it in. I won’t have the opportunity to test it for a while though.

straithe, 2023-11-28:
Please update us with the output of "hackrf_info" when you can test the device.

AlexLandherr, 2023-11-28:
> Please update us with the output of "hackrf_info" when you can test the device. 

Will do my best to remember that.

AlexLandherr, 2023-12-02:
> Can you please paste the output you get when you run "hackrf_info" on command line when your HackRF is plugged in?

```
alexl@PD70PNP:~$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc304422cf
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006a475f
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```

AlexLandherr, 2023-12-07:
> Everything looks fine there, you should be able to use `hackrf_clock -i` to check if the clock input is detected. If it is detected it will be used automatically, you do not need to do anything from gqrx.

I get this:
```
alexl@PD70PNP:~$ hackrf_clock -i
CLKIN status: no clock signal detected
```

which I think is because the Tiny TXCO isn't using the SMA connector labeled "CLK IN".

martinling, 2023-12-08:
The CLKIN signal is also present on the P22 header, not just on the SMA port. It should work fine through the header.

I suggest you raise this with NooElec. There were some problems back when the r9 hardware first came out, with TinyTCXO boards that were not producing enough signal voltage to meet the HackRF's specifications for the the clock input. Earlier revisions had been more forgiving.

We did let them know about this, and I know they were working on dealing with it, so they may have a newer version of the module that they can send out for you.

AlexLandherr, 2024-01-10:
> The CLKIN signal is also present on the P22 header, not just on the SMA port. It should work fine through the header.
>
> I suggest you raise this with NooElec. There were some problems back when the [...]

I contacted NooElec and today I got a new TCXO which seems to work based on the terminal output:
```
alexl@PD70PNP:~$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc304422cf
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006a475f
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
alexl@PD70PNP:~$ hackrf_clock -i
CLKIN status: clock signal detected
alexl@PD70PNP:~
```

martinling, 2024-01-10:
Great! I'm glad they were able to resolve that for you.

closed 2024-01-10 by martinling