greatscottgadgets/hackrf #1381: Frequency limits

lu8drh, 2023-11-27:
### What would you like to know?

Hello, I am starting a project to use Hackrf in the 10 Ghz band. I use a mixer and an external local oscillator. Using the GPIO pins, I activate both, but I want to tell Hackrf that its frequency limit is 10.5 Ghz and I can't find how to modify them. My question is, the frequency limits are in the firmware or in the drivers of the software used (SDR ANGEL in my case). Thanks, any help will be appreciated.

martinling, 2023-12-08:
You don't need to tell the HackRF that it's working with an external frequency converter. It doesn't know how to do anything with that information. You can either work out the offset manually, or tell SDRAngel about the converter. I'm not personally familiar with SDRAngel yet but I know it supports this, see [this issue](https://github.com/f4exb/sdrangel/issues/493) which has an explanation of how to do it.

lu8drh, 2023-12-08:
hello. thanks for answering. My intention is not to use the offset of the SDR Angel, so that way I move with agility from one band to another and using the GPIO pins of the HackRF, I turn on the transverter. To be clearer, when the HackRF receives a frequency between 10 and 10.5 Ghz, it turns on the OL and then sets itself to 10Ghz - OL. Switching between bands (2.4, 5.7 and 10 Ghz) is carried out using a multiple output relay.
[image]

martinling, 2023-12-08:
OK, I see what you mean.

Firstly note that we have an RF switching add-on for HackRF called [Opera Cake](https://greatscottgadgets.com/hackrf/operacake/), which can handle automatically switching between different signal paths based on the target frequency. If you haven't already selected your RF switches, you might want to consider using Opera Cake for that.

The software/firmware support for Opera Cake doesn't currently include the ability to apply a frequency offset for certain RF paths, but that is something that could be added if someone were to write the code, and I think it would be quite a useful feature. As well as converters for frequencies above 6GHz, it's quite common for people to want to use an external upconverter such as the Ham-It-Up for improved performance lower end of the spectrum.

lu8drh, 2023-12-08:
Yes, I know Opera, but it is not available here, so I use a 4-port Teledine relay. I built the external switching unit (TX/RX + BAND Swich) and have it working. I have the TX amplifiers for all bands (10W 2.4Ghz, 4W 5.7Ghz, 1W 10Ghz), the reception LNAs and the relays suitable for switching TX/RX for each band.
As part of my job, I have a small laboratory to work up to 26Ghz (SA, signal generator, power meter, etc.).
Right now, I am with the project in test mode. I already modified the HackRF firmware to work with my hardware, but I just want to see if it is possible to modify the frequency limits. The reason for this is that after it is working, I want to make specific software for it.

martinling, 2024-01-09:
The firmware will reject frequencies above 7250 MHz.

There's no frequency limits applied by libhackrf - it just passes the requested values to the firmware.

There are limits applied in the `hackrf_transfer` and `hackrf_sweep` command line tools. If you've already modified the firmware to handle a wider frequency range via controlling an external downconverter, then you can patch those tools to remove their internal limits.

For other software, limits might be found in whatever code is adapting between that software and libhackrf - e.g. the OsmoSDR or Soapy blocks.

closed 2024-02-13 by martinling