greatscottgadgets/hackrf #1388: Can't pick up any radio signal

generaI-kenobi, 2023-12-09:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

To see FM frequencies following the demo - `https://greatscottgadgets.com/sdr/`

### What outcome actually happened?

No frequencies shown.

### What operating systems are you seeing the problem on?

Ubuntu 22.04.3 LTS

### What is the output of ```hackrf_info```?

hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000453c64c8254f758f
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa004cb3c 0x0062435e

### Are you using any third-party software?

GNU Radio

### Are you using any third-party hardware?

N/A

ehuam, 2024-01-03:
What do you get when you run this in the command line

hackrf_sweep -f 2400:2490

straithe, 2024-03-14:
@generaI-kenobi have you updated to the latest firmware?

closed 2024-05-22 by straithe