greatscottgadgets/hackrf #1389: Can not recieve after transmitting

Amarthol78, 2023-12-16:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

To be able to receive after transmitting

### What outcome actually happened?

Need to turn the unit off then back on after transmitting to be able to receive.

### What operating systems are you seeing the problem on?

only tried on the portapack

### What is the output of ```hackrf_info```?

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc343375cf
Board ID Number: 4 (Unknown Board ID)
Firmware Version: n_231216 (API:1.08)
Part ID Number: 0xa000cb3c 0x00584768

### Are you using any third-party software?

n/a

### Are you using any third-party hardware?

portapack h2+

Amarthol78, 2023-12-24:
Ok, it seems that transmitting is not really working. I tried the BLE spam from the portapack to my phone but nothing ever showed up so I then tried other transmissions as well with an RTL-SDR plugged into my laptop, I see signal but there is no content.

And I still need to power cycle the Hackrf to be able to receive again. I have only had this for a couple weeks and its always done this, please, I could use some help!

M0rn1ng574r, 2023-12-25:
Have you chosen the type of phone you have? On BLE spam, If you chose Android but you have iOS this will not work!

Amarthol78, 2023-12-25:
Filed a bug report on the Mayhem github repo. Turns out I had to refresh the CPLD, it's all good and working now! Thanks for the help.

closed 2023-12-25 by Amarthol78